MRPT/mrpt #118: Integrate MRPT with MATLAB (mex files)

jlblancoc, 2015-01-07:
@jesusbriales @jlblancoc Please, add here new tasks as we think of them, so they don't get lost! (Just click on the check to mark/unmark them when done. See also https://github.com/jlblancoc/mrpt/milestones )

List of pending tasks in this line:
- [ ] Create `#define` macros to help automating the serialization to mxArray format (Is it worth?)
- [ ] Create a doxygen page (in `doc/doxygen-pages`) mentioning all provided mex scripts. (Should we document them there, in doxygen? I think it's a good option...)
- [ ] Think & test where compiled .mex* files should go after `install`, thinking of Windows package installers (`Program Files/MRPT/bin`, `Program Files/MRPT/mex`,..??)

Finally:
- [ ] Merge branch new-mex-lib

jlblancoc, 2015-07-03:
@jesusbriales  I just updated this to block mrpt 1.3.2, so we can release 1.3.1 sooner, ok? ;-)

jesusbriales, 2015-07-04:
Ok! :)

closed 2026-10-02 by jlblancoc