MRPT/mrpt #129: on windows 8 + mrpt 1.3.0: webcams don't work in camera-calib.exe

gitunit, 2015-01-23:
i've tried two webcams, one notebook built-in cam and one usb-cam.
these cameras don't work with the new version.
however, both cameras work flawlessly with version mrpt 1.2.2

jlblancoc, 2015-01-23:
My guess is that you refer to the precompiled (Windows installer) mrpt version, right? 
In that case, the problem is *probably* related to mrpt shipping opencv 3.0.0 beta instead of the previous 2.8.x. 

The best way to find out more is if you try to build mrpt from sources against opencv 2.8.x and check again...

gitunit, 2015-01-24:
yes, its the precomiled version. i will see if i can recompile it with another opencv version. i have 2.4.9 on my machine, will that work with mrpt?

jlblancoc, 2015-01-24:
Sure

jlblancoc, 2015-02-15:
Will close this, unless it's confirmed that the problem is mrpt-specific.

closed 2015-02-15 by jlblancoc