MRPT/mrpt #134: Problem to use CSimplePointsMap extractPoints() function

anthonykylai, 2015-02-15:
I try to use the function extractPoints() of CSimplePointsMap as the code is as follow:

		mrpt::slam::CSimplePointsMap theMap;
		mrpt::slam::CSimplePointsMap theMap2;

		theMap.extractPoints(TPoint3D(0,0,0),TPoint3D(1,1,0),theMap2,1,1,1);
		theMap.insertionOptions.minDistBetweenLaserPoints	= 0.1;
	
But it seems the complier generate the following error code:
\test.cpp(66) : error C2664: 'mrpt::slam::CPointsMap::extractPoints' : cannot convert parameter 3 from 'mrpt::slam::CSimplePointsMap' to 'mrpt::slam::CPointsMap *'

is anyone can help me to solve this problem. Thanks!

jlblancoc, 2015-02-15:
As you say: 'mrpt::slam::CPointsMap *' is expected, so you have to add an & in front of the variable: 

    theMap.extractPoints(TPoint3D(0,0,0),TPoint3D(1,1,0),&theMap2 );

closed 2015-02-15 by jlblancoc

anthonykylai, 2015-02-16:
Dear Jlblancoc, it works! Thank you!