MRPT/mrpt #136: Calling ::system with user-supplied strings

jpgr87, 2015-03-01:
The CImpinjRFID::startDriver() function calls ```system``` with an argument constructed from user-supplied strings (coming from an ini file).  This functionality can be hijacked to force mrpt to run arbitrary commands.  You could mitigate this by checking ```driver_path``` to ensure it's a real path, and then sanitizing the rest of the arguments (making sure ```port``` is an integer, ```IPm``` is a valid IP address, and ```reader_name``` is alphanumeric or free of shell constructs like ;, &&, ||, etc.

On the other hand, since the ```driver_path``` is a user-defined value it can be set to any malicious binary or shell script anyway.  It may not be worth the effort to sanitize the arguments if you're running arbitrary commands to begin with.

It looks like this is the only dangerous call to system in mrpt 1.3.0.

jlblancoc, 2015-03-01:
Thanks! Sure, I'll look at this.

jlblancoc, 2015-03-02:
After reviewing the inner workings of  CImpinjRFID, I think it would not be worth checking that command: it's not something that will arrive from a remote computer, etc. so it makes sense to me that the user-provided command is his/her responsibility.

One small improvement would be checking it is an executable without additional arguments or shell constructors as you said, but... again, what if the user really pretend to exploit that possibility? 

So, I will close the issue without changes if no better ideas arrive.

Thanks a lot anyway for the feedback.

closed 2015-03-02 by jlblancoc