MRPT/mrpt #139: No such file or directory for HOKUYO sample

droter, 2015-04-03:
Hi,

I am trying to run the HOKUYO sample and I am getting a no such file or directory error.

jh (~/mrpt/mrpt/samples/HOKUYO_laser_test): make
Scanning dependencies of target HOKUYO_laser_test
[100%] Building CXX object CMakeFiles/HOKUYO_laser_test.dir/test.o
/home/jh/mrpt/mrpt/samples/HOKUYO_laser_test/test.cpp:13:46: fatal error: mrpt/obs/CObservation2DRangeScan.h: No such file or directory
 #include <mrpt/obs/CObservation2DRangeScan.h>
                                              ^
compilation terminated.
make[2]: *** [CMakeFiles/HOKUYO_laser_test.dir/test.o] Error 1
make[1]: *** [CMakeFiles/HOKUYO_laser_test.dir/all] Error 2
make: *** [all] Error 2

Thanks for your help,

Matt

jlblancoc, 2015-04-03:
Hi, 

As some headers and namespaces changed in recent releases, make sure to be building against MRPT 1.3.0 or newer. Otherwise, download the sources for your mrpt version and use the corresponding example code from `samples/NAME/*`. 

I will close the issue in the hope this solves the problem. 

Cheers.

closed 2015-04-03 by jlblancoc

droter, 2015-04-06:
Hi,

Would these questions be better on the google group site?

I downloaded the source and compiled.  

Now I am getting this error:
/usr/bin/ld: cannot find -lmrpt-hwdrivers

:~/mrpt/samples/HOKUYO_laser_test$ make
Linking CXX executable HOKUYO_laser_test
/usr/bin/ld: cannot find -lmrpt-hwdrivers
collect2: ld returned 1 exit status

Matt