MRPT/mrpt #148: Pose composition with CPose3DQuat

KDemeul, 2015-06-25:
Our team discovered what seems to be a bug in the pose composition using CPose3DQuat.
Here is a quick summary of the situation:
When using the pose composition with a CPose3D object and the pose composition using a CPose3DQuat, the resulting positions are differents. It is visible when trying to create a circle: We start in pose (0,0,0) with angles (0,0,0) and start rotating around the y-axis by an angle of 4°. After 90 iterations (meaning that we have turned 360°), the robot should be back at its original pose. It is the case with the CPose3D, but there is an offset with the CPose3DQuat.

Here is a quick code to reproduce the problem: 

```
// First, let's create a gui to visualize the problem
bool enableVisualisation = true;
mrpt::gui::CDisplayWindowPlots windowPlot("Window");;
windowPlot.hold_on();
// Create a identity matrix 2x2 
// (used with windowPlot.plotEllipse(...) only, its not relevant to the problem)
mrpt::math::CMatrixDouble22 cov;
cov.setIdentity();

// Here, we create two actions: one using CPose3D, the other using CPose3DQuat
// Both actions are based on the same rigid body transformation:
// A rotation of theta around the y-axis with a translation of the robot
double theta = mrpt::math::wrapToPi(-M_PI / 180. * 4.0);
double r = 1;
// Create the rigid body transformation
mrpt::math::CMatrixDouble44 RT;
RT << cos(theta), 0, sin(theta), r * sin(theta),
    0, 1, 0, 0,
    -sin(theta), 0, cos(theta), -r * (cos(theta) - 1),
    0, 0, 0, 1;
mrpt::poses::CPose3D action(RT);
mrpt::poses::CPose3DQuat actionQuat(RT);

// We will now compose this actions 90 times,
// meaning that we will rotate around the y-axis with an angle 90 * 4° = 360°.
mrpt::poses::CPose3D robotPose;
mrpt::poses::CPose3DQuat robotPoseQuat;
for (size_t t = 0; t < 90; t++){
    robotPose += action;
    robotPoseQuat += actionQuat;
    if (enableVisualisation){
        windowPlot.plotEllipse(robotPose.x(), robotPose.z(), cov, 0.002, "b.3");
        windowPlot.plotEllipse(robotPoseQuat.x(), robotPoseQuat.z(), cov, 0.002, "r.3");
    }
}

// We now cast the CPose3D in CPose3DQuat
mrpt::poses::CPose3DQuat robot_pose_transformed(robotPose);

// As you can see here, the two positions are different!
std::cout << "robotPoseQuat: " << robotPoseQuat << std::endl;
std::cout << "robotPosetransformed: " << robot_pose_transformed << std::endl;

// Display the problem: in blue, the circle using CPose3D, in red, the circle using CPose3DQuat
windowPlot.hold_off();
windowPlot.waitForKey();
```

KDemeul, 2015-06-25:
Something that might be helpfull: 
- The pose composition using CPose3DQuatPDFGaussian and CPose3DPDFGaussian leads to the same resuls (therefore, it seems that the bug is only present when using CPose3DQuat)

jlblancoc, 2015-06-25:
@KDemeul  Again, thanks for reporting this weird bug! It's solved, refer to the commits. Cheers.

commit 821c0118 (2015-06-25): BUG FIX related to chaining quaternion transformations (Fixes #148)

- Fix wrong quaternion cross product when target variable is one of the
operands.
- Also affected the += operator of mrpt::poses::CPose3DQuat
- Added a bunch of new unit tests.
closed 2015-06-25 by jlblancoc