MRPT/mrpt #156: kinect-3d-slam seems to not work as intendet

ensonic, 2015-07-08:
Initially I see "LOST! Please press 'r' to restart". Pressing 'r' does not make the message go away, but leaving the camera for some time make the message go away. When I then move very slowly I get still some initial point cloud residue on the screen. Now when I keep moving the point cloud expands on one side, but disappears on the other side.

Finally I can close the window, but the 'esc' key that is mentioned on the text on the top left does not work.

closed 2015-07-13 by jlblancoc

ensonic, 2015-07-15:
Hmm, I am now gettting a bunch of

    [se3_l2_robust] maximum size is == 0!

lines, but other than that it does not work better. I am rotating the kinect super slowly on my desk, but I get the "LOST! Please press 'r' to restart" within a few degrees.

jlblancoc, 2015-07-18:
I'm afraid it has to do with parameter tuning in the KLT feature tracker, etc. This app was just a quick test, but I left it published as an example. Perhaps it should be better to use other more robust feature trackers out there, from OpenCV, ...

Cheers!

ensonic, 2015-07-20:
Thanks for the fixes anyway. I was simply looking for SLAM demo, since I've got a used kinect on the flee-market. I'll look for some other demos.

commit f4364508 (2015-07-13): Fix a bug in mrpt::tfest::se3_l2_robust()

Fix a bug in mrpt::tfest::se3_l2_robust() that led to it returning
without trying to find a good consensus solution.
It also affected the demo app kinect-3d-slam (Fixes #156)
Added new unit test to cover this function.