MRPT/mrpt #160: wxWidgets was not found automatically. Please, set wxWidgets_ROOT_DIR to the lib directory to enable it in MRPT. How should I fix it?

5aou, 2015-07-16:
Hi all, I encounter this error when I do : ccmake .
wxWidgets was not found automatically. Please, set wxWidgets_ROOT_DIR to the lib directory to enable it in MRPT. Can you tell me please how should I fix it? thank you a lot.

Using : linux mint.

jlblancoc, 2015-07-16:
I guess you should install the -dev or -devel package equivalent for
wxWidgets in Linux mint, and then CMake should find it automatically...

JL

5aou, 2015-07-16:
Ok thank you, it works now.

5aou, 2015-07-16:
re, I want to implement a kf-slam to have results like those : https://www.youtube.com/watch?v=eOkzd_i_-ps  (camera poses), but the application page : http://www.mrpt.org/Application:kf-slam/ gives me an error. Can you give me please a tutorial or another page where I can find the application.... thank you a lot.

closed 2015-07-17 by jlblancoc