MRPT/mrpt #171: [mrpt::synch] Create lightweight (non-Boost) shared/unique/upgradable (read/write) locks

jlblancoc, 2015-11-26:
See: http://stackoverflow.com/a/244376/1631514

jolting, 2016-02-15:
unique locks are already supported by c++11
http://en.cppreference.com/w/cpp/thread/unique_lock

shared locks are supported by c++14 compilers. This might be a problem until c++14 is ubiquitous.
http://en.cppreference.com/w/cpp/thread/shared_lock

Unfortunately, upgradeable locks aren't supported, but
It's worth reading:
https://isocpp.org/blog/2013/04/n3659-shared-locking
http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2013/n3568.html

Here is Howard's public domain version of an upgradeable lock if you need one.
https://github.com/HowardHinnant/upgrade_mutex

jlblancoc, 2016-02-15:
Again, thanks for the pointers!
The creation of non-C++11, non-boost classes in MRPT is intended to serve as a workaround until we definitively enforce the use of C++11/14 in MRPT 2.0.0 (See also: issue #66 )

jlblancoc, 2016-07-15:
Since the current codebase does not make use of these features, I think that it makes more sense to just drop this topic for "milestone 1.5.0" and just use native C++11/14 classes in "milestone 2.0.0". 

So: closing this ticket.

closed 2016-07-15 by jlblancoc

jolting, 2016-07-15:
Agreed