MRPT/mrpt #174: TP-Space navigation: include info for closest obstacles not only collisions

jlblancoc, 2015-12-09:
Try to include that info into precomputed tables: it may take more memory, but may allow robots to navigate more correctly finding the middle of gaps (safer navigation).

jlblancoc, 2016-03-03:
More details on what is to be done here: 
- [ ] Update mrpt::nav::build_PTG_collision_grids() and associated data structures to add a second LUT (look-up-table), not for collisions, but for the **minimum clearance** between any part of the vehicle (which may be of any shape) and any "obstacle-cell" in a grid (x,y), for any `alpha` (TP-Space path parameter) and for a small set of distancies along each trajectory. Instead of a 4-dimensional LUT, it may be more efficient to reuse MRPT 2D grids (just like the current collision LUT) storing, at each cell, a `CMatrix` with rows and cols being the `alpha` and `distance` index.

[image]

- [ ] Update (de)serialization so the new LUT is also loaded/saved to disk and needs to be computed only once.

- [ ] Create a new method (in `mrpt::nav::CParameterizedTrajectoryGenerator`?) to evaluate the minimum clearance of a path given its `alpha` and its `distance` along the path. Clearances may have to be interpolated somehow between the small discrete set of distances for which clearance was precomputed. 

- [ ] Make the clearance accesible as a new parameter to the reactive navigation engine, and decide good default weights so the robot tends to avoid paths too close to obstacles, even if theoretically collision-free.

chatrasen, 2016-03-07:
@jlblancoc 
In the previous LUT we are storing the (a,d) which make the robot land in that cell starting from the **position (0,0)** right?

And in the new LUT that we want to design, we will have to simulate trajectories from the cell we are storing information in and then find the minimum clearance? Is that what we have to do? If wrong, please correct me.

jlblancoc, 2016-03-07:
The previous LUT is built in [`mrpt::nav::build_PTG_collision_grids()`](https://github.com/MRPT/mrpt/blob/master/libs/nav/src/tpspace/motion_planning_utils.cpp#L26). 
In [this code](https://github.com/MRPT/mrpt/blob/master/libs/nav/src/tpspace/motion_planning_utils.cpp#L113) it can be seen how an entry is added to the LUT only when the **shape of the robot** collides with the cell... which may be quite different that the robot origin (the local 0,0 coordinats on the vehicle) reaches that cell ;-)

The new code might be incorporated inside `build_PTG_collision_grids()` as well, probably reusing all the outer loops. The missing part is evaluating the **minimum distance** from (obs_x,obs_y) to the robot polygon. 
May need to use geometry functions in `mrpt::math`.

chatrasen, 2016-03-07:
@jlblancoc 
Thanks, got my first doubt cleared. 

For the new LUT, what you mean to say is, inside the same if as you mentioned in the above code, store the minimum distance from (obs_x, obs_y) corresponding to the (a,d) pair and keep rest of the code as it is, right?

dheerajgattupalli, 2016-03-07:
@chatrasen  I dont think we can write it inside the if condition...

For each obstacle point(i.e npoint) we have to compute the distance from all points on the robot polygon to obstacle point and save the minimum a,d.. so this part should be written outside the if condition..
@jlblancoc 
correct me if i am wrong anywhere

jlblancoc, 2016-03-07:
@dheerajgattupalli is right, the new code should go outside of the `if`
However, note that the closest distance may lie in the middle of a segment, so checking for vertices should not be enough. Hint: [`TSegment2D::distance()`](http://reference.mrpt.org/devel/structmrpt_1_1math_1_1_t_segment2_d.html#a7c679020f08ff96e8067fe3232b04aba)

Also, remember that to avoid excessive memory usage, clearance should not be evaluated for all `d`  values (`n` for loop)...

chatrasen, 2016-03-07:
@dheerajgattupalli why are the npoints obstacle points? They have just been simulated from the trajectory.

dheerajgattupalli, 2016-03-07:
@chatrasen true npoints are points from trajectory not obstacle points ... i edited my comment :-)

jlblancoc, 2016-03-09:
Thank you @chatrasen for your great contribution in Pull request #202 !! 
I just marked as done the first 3 checkboxes above. However, there are some small issues that should improved as added in code comments in the PR. It would be great if you or someone else could go on with this line, otherwise it will be addressed in the future, no worries...

chatrasen, 2016-03-10:
@jlblancoc 
The CReactiveNavigationSystem::STEP3_WSpaceToTPSpace() function you mentioned takes in one parameter as the vector<float> of obstacles. To incorporate the collison grid, should I use a pair inside the vector or create a new vector for this purpose. Because using a pair will change the code a lot for the existing users I guess though it is convinient for a new user

jlblancoc, 2016-03-12:
Good point @chatrasen ! 
Indeed, that part must be changed... I guess that it should be worth changing it to make it "right" and more maintainable in the future: one new struct could be created (instead of a pair, because we have a pair for now, but who knows in the future?) and use a vector of that struct.

chatrasen, 2016-03-12:
Yea right, this would be take care of any possible changes in future. Will be working on this soon. Presently working on the GSoC apllication :P

jlblancoc, 2016-03-12:
Sure!  First is first ;-)

PR #409 (2016-12-29): PTG: added clearance computation
This closes #174. 
The old experimental branch can be now deleted.
closed 2016-12-29 by jlblancoc