MRPT/mrpt #184: Build python pymrpt Debian/Ubuntu packages

jlblancoc, 2016-02-01:
TO-DO for a next release:
See: https://wiki.debian.org/Python/LibraryStyleGuide
Update: https://github.com/MRPT/mrpt/wiki/PythonBindings

jlblancoc, 2018-06-12:
Update: Since today, `pymrpt` is released within the ROS package [`mrpt1`](http://wiki.ros.org/mrpt1) or [`mrpt2`](http://wiki.ros.org/mrpt2): 

    sudo apt-get install ros-${ROS_DISTRO}-mrpt1

so that's an alternative to quickly installing MRPT and its Python bindings.

PymZoR, 2019-02-15:
Hi @jlblancoc, I'm unable to install / use the `pymrpt` bindings. Installed it through `mrpt1`, but no evidence of any `pymrpt` files installed on my system. Output of  ``` dpkg-query -L ros-melodic-mrpt1 | grep py``` yields nothing.

jlblancoc, 2019-02-15:
Hi, 
You're right! Something happened at some time and we now don't have them built (look for "pymrpt"):
http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__mrpt1__ubuntu_bionic_amd64__binary/10/consoleFull

It looks like a problem of missing dependencies (boost python wrapper libs), since only these deps are installed (from the log above):

```
20:16:31 Invoking 'apt-get install -q -y -o Debug::pkgProblemResolver=yes apt-src cmake debhelper freeglut3-dev libassimp-dev libavcodec-dev libavformat-dev libavutil-dev libeigen3-dev libjpeg-dev libopencv-dev libpcap0.8-dev libsuitesparse-dev libswscale-dev libudev-dev libusb-1.0-0-dev libwxgtk3.0-dev ros-melodic-octomap wx-common zlib1g-dev'
``` 
Pull requests are welcome to fix it! :-) 
On the other hand, providing a real pymrpt Debian package by updating packaging/debian/* would be also great, so they would be served directly from the PPA instead of the ROS builds....

jlblancoc, 2023-07-03:
The package python3-pymrpt is finally available, yay!

PR #874 (2018-12-03): Fix CMake targets in Debian pkgs
Closes: #870 
Closes: #184
commit 4908c90f (2018-12-03): Fix CMake targets in Debian pkgs (#874)

Closes: #870
Closes: #184
closed 2023-07-03 by jlblancoc