MRPT/mrpt #186: Compile error...

roliveiradoteu, 2016-02-05:
Having issues compiling the master branch in Gentoo:

`[ 38%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CWirelessPowerGridMap2D.cpp.o
In file included from /usr/include/eigen3/Eigen/SparseCholesky:26:0,
                 from /home/ro/mrpt-git/libs/maps/include/mrpt/maps/CRandomFieldGridMap2D.h:25,
                 from /home/ro/mrpt-git/libs/maps/include/mrpt/maps/CWirelessPowerGridMap2D.h:13,
                 from /home/ro/mrpt-git/libs/maps/src/maps/CWirelessPowerGridMap2D.cpp:12:
/usr/include/eigen3/Eigen/src/SparseCholesky/SimplicialCholesky.h: In member function 'void Eigen::SimplicialCholeskyBase<Derived>::ordering(const MatrixType&, Eigen::SimplicialCholeskyBase<Derived>::CholMatrixType&)':
/usr/include/eigen3/Eigen/src/SparseCholesky/SimplicialCholesky.h:684:5: error: 'minimum_degree_ordering' is not a member of 'Eigen::internal'
     internal::minimum_degree_ordering(C, m_Pinv);
     ^
libs/maps/CMakeFiles/mrpt-maps.dir/build.make:158: recipe for target 'libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CWirelessPowerGridMap2D.cpp.o' failed
make[2]: *** [libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CWirelessPowerGridMap2D.cpp.o] Error 1
CMakeFiles/Makefile2:727: recipe for target 'libs/maps/CMakeFiles/mrpt-maps.dir/all' failed
make[1]: *** [libs/maps/CMakeFiles/mrpt-maps.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
`

jlblancoc, 2016-02-06:
Hi, 

It seems to be due to a bug in an old version of Eigen, see: https://forum.kde.org/viewtopic.php?f=74&t=110374

From MRPT's CMake, check the variable `EIGEN_USE_EMBEDDED_VERSION` which hopefully is a newer version. Otherwise, try installing under `/usr/local` a new version of Eigen. 

If none of this works, please, report your complete MRPT CMake config with:

    VERBOSE=1 cmake . 

I'll close the issue, please reopen it if needed.

closed 2016-02-06 by jlblancoc

roliveiradoteu, 2016-02-07:
Upgraded eigen from 3.1.3 to 3.2.7 and compiled successfully. 

Thanks

Ricardo