MRPT/mrpt #197: Merge PbMap

jlblancoc, 2016-03-03:
From @EduFdez fork.
TO DO: 
- [ ] Changelog: Describe what's new in PbMap

EduFdez, 2016-03-03:
I'm working on this, I didn't do the pull request before because some things were not stable yet, but I'm planning to do it soon!

jlblancoc, 2016-03-29:
Hi @EduFdez , 
I intend to release 1.4.0 soon, so I'll move this to 1.4.1 for you to have plenty of more time for testing, ok? Cheers! ;-)

jlblancoc, 2017-01-26:
Hi @EduFdez : Any news on this pull-request? 
Shall I do the pull-request with the code in your branch as is? Is it stable?

Saludos!

jlblancoc, 2018-02-14:
Hi @EduFdez !
I'm closing this one for now. If you want to do the merge in the future, I guess it would require doing a git rebase and/or substantial manual conflict solving... let us know if that's the case and you need a hand.

closed 2018-02-14 by jlblancoc