MRPT/mrpt #211: Missing '-dbg' suffixed libraries in libmrpt-dbg on Ubuntu

jolting, 2016-03-24:
```
/usr/bin/ld: cannot find -lmrpt-base-dbg
/usr/bin/ld: cannot find -lmrpt-opengl-dbg
/usr/bin/ld: cannot find -lmrpt-obs-dbg
/usr/bin/ld: cannot find -lmrpt-maps-dbg
/usr/bin/ld: cannot find -lmrpt-gui-dbg
/usr/bin/ld: cannot find -lmrpt-vision-dbg
/usr/bin/ld: cannot find -lmrpt-hwdrivers-dbg
/usr/bin/ld: cannot find -lmrpt-tfest-dbg
/usr/bin/ld: cannot find -lmrpt-slam-dbg
/usr/bin/ld: cannot find -lmrpt-nav-dbg
```
https://github.com/MRPT/mrpt/blob/55b997e9076d522924f03eed6944b53eb2657f3b/parse-files/MRPTConfig.cmake.in#L185

I think setting  MRPT_DONT_USE_DBG_LIBS before calling find package is the right thing to do when trying to debug your app. It appears that libmrpt-dbg doesn't really install any debug libraries. Instead it just installs debug information in a place where gdb can find it.

zazics, 2016-03-24:
Same problem. My solution was to build from source code with "DEBUG" mode. 
Tested on Odroid XU-4 and debugging finally works  (with some minors corrections in code).

jlblancoc, 2016-03-26:
In the past we provided them in the `-dbg` package, but we had to change to comply with the Debian standards:
https://wiki.debian.org/DebugPackage

However, it may be possible to use another name to provide prebuilt debug libs. In the meanwhile, yes, setting `MRPT_DONT_USE_DBG_LIBS` is the quickest workaround!

closed 2016-05-11 by jlblancoc