MRPT/mrpt #224: rawlog-grabber output log "peek: seek out of range", when grabber the NMEA data  of GPS

9578577, 2016-04-11:
Today , I found the rawlog-grabber always print the information "peek: seek out of range" in log.
when I use the rawlog-grabber to grabber the NMEA data of GPS.
The information would be shown when the rawlog file will more than 18MB.

So, I change the buffer size of GPS serial port to 0x20000, and modify the 48th line of parser_NMEA file
from "if (val=='\r' || val=='\n') {" to "if (val=='\n') {"

Now, It is normal.

jlblancoc, 2016-04-11:
Thanks!
But that seems sort of a hack / workaround... 

I will investigate the issue in depth adding intensive unit tests around `peek()` and will notify the results here.

jlblancoc, 2016-04-12:
@9578577 please, test it now after the lastest commit. It should be fixed...

commit e8601c78 (2016-04-12): Fix false positive error detection in circular_buffer::peek() (Fix #224 )
closed 2016-04-12 by jlblancoc