MRPT/mrpt #225: The Eigen/Array header does no longer exist in Eigen3

prpsymbol, 2016-04-12:
Hi all, I am trying to compile mrpt 1.3.2 under Windows and MinGW32.  The CMake configuration has the EIGEN_USE_EMBEDDED_VERSION checked, and my make command is [mingw32-make -f makefile CXXFLAGS="-std=gnu++11" SHARED=1 UNICODE=1 BUILD=release]

The build explodes compiling bayes/CParticleFilter.cpp, and yields the error message "The Eigen/Array header does no longer exist in Eigen3. All that functionality has moved to Eigen/Core."  

What am I doing wrong (besides building under Windows ;) )?

jlblancoc, 2016-04-12:
Hi! I think that's already fixed in the `master` branch in GitHub... please, give it a try and re-open the issue if it's not. 

Best,

prpsymbol, 2016-04-12:
Hi Jose,

I just ran the build from a fresh check out from GIT, and I get the same error on the same file. How do I reopen the issue? Or should I just create an identical new one?

-Paul

jlblancoc, 2016-04-12:
Strange... I'm reopening it for you. 
Please, could you attach the build log so I can follow the track the #include's leading to the error?

prpsymbol, 2016-04-12:
[BuildLog.txt](https://github.com/MRPT/mrpt/files/215763/BuildLog.txt)

jlblancoc, 2016-04-12:
Look at the sequence of #includes: it seems Eigen/Array is mistakenly trying to be included from GCC C++11 STL tuple!! Perhaps is a problem related to Windows & case insensitivity... 

Quick solution: remove the file Eigen/Array !!
If you want to investigate it a bit more: I think that only `MRPT\otherlibs\eigen3` should be included with `-I` to GCC, so can't understand why `#include <array>` actually finds `#include <Eigen/Array>`... if possible, you could run mingw make with the environment variable VERBOSE=1 (`set VERBOSE=1` in DOS) to see the full list of `-I` arguments... 

In file included from C:/Qt/Tools/mingw492_32/i686-w64-mingw32/include/c++/tuple:39:0,
                 from C:/Qt/Tools/mingw492_32/i686-w64-mingw32/include/c++/functional:55,
                 from C:/Users/e815228/Documents/MRPT-SVN/otherlibs/eigen3/Eigen/Core:153,
                 from C:/Users/e815228/Documents/MRPT-SVN/otherlibs/eigen3/Eigen/Dense:1,
                 from C:/Users/e815228/Documents/MRPT-SVN/libs/base/include/mrpt/utils/types_math.h:31,
                 from C:/Users/e815228/Documents/MRPT-SVN/libs/base/include/mrpt/math/CMatrixTemplateNumeric.h:13,
                 from C:/Users/e815228/Documents/MRPT-SVN/libs/base/include/mrpt/random/RandomGenerators.h:13,
                 from C:/Users/e815228/Documents/MRPT-SVN/libs/base/include/mrpt/random.h:13,
                 from C:\Users\e815228\Documents\MRPT-SVN\libs\base\src\bayes\CParticleFilterCapable.cpp:13:
C:/Users/e815228/Documents/MRPT-SVN/otherlibs/eigen3/Eigen/array:8:4: error: #error The Eigen/Array header does no longer exist in Eigen3. All that functionality has moved to Eigen/Core.
   #error The Eigen/Array header does no longer exist in Eigen3. All that functionality has moved to Eigen/Core.

Anyway, I'll close since it **seems** not to be general problem of MRPT...

prpsymbol, 2016-04-12:
[BuildLog.txt](https://github.com/MRPT/mrpt/files/215960/BuildLog.txt)

Here is the verbose output. If the problem is case sensitivity, what files/directories are being confused?

prpsymbol, 2016-04-12:
[BuildLog.txt](https://github.com/MRPT/mrpt/files/215975/BuildLog.txt)

Last log was with 'verbose' in the mingw23-make command, this one is with the DOS variable VERBOSE set to 1.

jlblancoc, 2016-04-12:
The last buildlog was the key... I found a misplaced -isystem flag which, apparently, went unnoticed until now since in Linux, the case sensitivity of the filesystem shadowed the error. 
Thanks a lot for the feedback!

Try this patch and hopefully it's solved:
https://github.com/MRPT/mrpt/commit/89a778edcbf66b3546e691b895f3ea4f86d5addd

prpsymbol, 2016-04-13:
Thanks! That cured this problem… only to expose the next. Build log attached.

89a778e<https://github.com/MRPT/mrpt/commit/89a778edcbf66b3546e691b895f3ea4f86d5addd>

"C:\Program Files (x86)\CMake\bin\cmake.exe" -HC:\Users\e815228\Documents\MRPT-SVN -BC:\Users\e815228\Documents\MRPTBuild --check-build-system CMakeFiles\Makefile.cmake 0
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E cmake_progress_start C:\Users\e815228\Documents\MRPTBuild\CMakeFiles C:\Users\e815228\Documents\MRPTBuild\CMakeFiles\progress.marks
mingw32-make -f CMakeFiles\Makefile2 all
mingw32-make -f otherlibs\assimp\contrib\zlib\CMakeFiles\zlibstatic.dir\build.make otherlibs/assimp/contrib/zlib/CMakeFiles/zlibstatic.dir/depend
mingw32-make -f otherlibs\wxThings\CMakeFiles\mrptwxthings.dir\build.make otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/depend
mingw32-make -f libs\base\CMakeFiles\mrpt-base.dir\build.make libs/base/CMakeFiles/mrpt-base.dir/depend
mingw32-make -f otherlibs\gtest-1.6.0\CMakeFiles\mrptgtest.dir\build.make otherlibs/gtest-1.6.0/CMakeFiles/mrptgtest.dir/depend
mingw32-make[2]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E cmake_depends "MinGW Makefiles" C:\Users\e815228\Documents\MRPT-SVN C:\Users\e815228\Documents\MRPT-SVN\otherlibs\gtest-1.6.0 C:\Users\e815228\Documents\MRPTBuild C:\Users\e815228\Documents\MRPTBuild\otherlibs\gtest-1.6.0 C:\Users\e815228\Documents\MRPTBuild\otherlibs\gtest-1.6.0\CMakeFiles\mrptgtest.dir\DependInfo.cmake --color=
mingw32-make[2]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make -f otherlibs\gtest-1.6.0\CMakeFiles\mrptgtest.dir\build.make otherlibs/gtest-1.6.0/CMakeFiles/mrptgtest.dir/build
mingw32-make[2]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E cmake_depends "MinGW Makefiles" C:\Users\e815228\Documents\MRPT-SVN C:\Users\e815228\Documents\MRPT-SVN\otherlibs\assimp\contrib\zlib C:\Users\e815228\Documents\MRPTBuild C:\Users\e815228\Documents\MRPTBuild\otherlibs\assimp\contrib\zlib C:\Users\e815228\Documents\MRPTBuild\otherlibs\assimp\contrib\zlib\CMakeFiles\zlibstatic.dir\DependInfo.cmake --color=
mingw32-make[2]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make -f otherlibs\assimp\contrib\zlib\CMakeFiles\zlibstatic.dir\build.make otherlibs/assimp/contrib/zlib/CMakeFiles/zlibstatic.dir/build
mingw32-make[2]: Nothing to be done for 'otherlibs/gtest-1.6.0/CMakeFiles/mrptgtest.dir/build'.
mingw32-make[2]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E cmake_depends "MinGW Makefiles" C:\Users\e815228\Documents\MRPT-SVN C:\Users\e815228\Documents\MRPT-SVN\otherlibs\wxThings C:\Users\e815228\Documents\MRPTBuild C:\Users\e815228\Documents\MRPTBuild\otherlibs\wxThings C:\Users\e815228\Documents\MRPTBuild\otherlibs\wxThings\CMakeFiles\mrptwxthings.dir\DependInfo.cmake --color=
mingw32-make[2]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make -f otherlibs\wxThings\CMakeFiles\mrptwxthings.dir\build.make otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/build
mingw32-make[1]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
[  1%] Built target mrptgtest
mingw32-make[1]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make[2]: Nothing to be done for 'otherlibs/assimp/contrib/zlib/CMakeFiles/zlibstatic.dir/build'.
mingw32-make[1]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
[  3%] Built target zlibstatic
mingw32-make[1]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make -f otherlibs\assimp\code\CMakeFiles\assimp.dir\build.make otherlibs/assimp/code/CMakeFiles/assimp.dir/depend
mingw32-make[2]: Nothing to be done for 'otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/build'.
mingw32-make[1]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
[  3%] Built target mrptwxthings
mingw32-make[1]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make[2]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E cmake_depends "MinGW Makefiles" C:\Users\e815228\Documents\MRPT-SVN C:\Users\e815228\Documents\MRPT-SVN\otherlibs\assimp\code C:\Users\e815228\Documents\MRPTBuild C:\Users\e815228\Documents\MRPTBuild\otherlibs\assimp\code C:\Users\e815228\Documents\MRPTBuild\otherlibs\assimp\code\CMakeFiles\assimp.dir\DependInfo.cmake --color=
mingw32-make[2]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make -f otherlibs\assimp\code\CMakeFiles\assimp.dir\build.make otherlibs/assimp/code/CMakeFiles/assimp.dir/build
mingw32-make[2]: Nothing to be done for 'otherlibs/assimp/code/CMakeFiles/assimp.dir/build'.
mingw32-make[1]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
[ 15%] Built target assimp
mingw32-make[1]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
mingw32-make[2]: Entering directory 'C:/Users/e815228/Documents/MRPTBuild'
"C:\Program Files (x86)\CMake\bin\cmake.exe" -E cmake_depends "MinGW Makefiles" C:\Users\e815228\Documents\MRPT-SVN C:\Users\e815228\Documents\MRPT-SVN\libs\base C:\Users\e815228\Documents\MRPTBuild C:\Users\e815228\Documents\MRPTBuild\libs\base C:\Users\e815228\Documents\MRPTBuild\libs\base\CMakeFiles\mrpt-base.dir\DependInfo.cmake --color=
mingw32-make[2]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
[... cut ...] sse3 -mssse3 -msse4.1 -msse4.2 -msse4a  -O3 -DNDEBUG   -o CMakeFiles\mrpt-base.dir\src\utils\CFileStream.cpp.obj -c C:\Users\e815228\Documents\MRPT-SVN\libs\base\src\utils\CFileStream.cpp
mingw32-make[2]: Leaving directory 'C:/Users/e815228/Documents/MRPTBuild'
CMakeFiles\Makefile2:389: recipe for target 'libs/base/CMakeFiles/mrpt-base.dir/all' failed
mingw32-make[1]: *** [libs/base/CMakeFiles/mrpt-base.dir/all] Error 2
makefile:148: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

jlblancoc, 2016-04-13:
(Reopening and changing title)

prpsymbol, 2016-04-14:
Hi Jose, thanks for the changes. I checked out a fresh version from the development trunk, and the build completed successfully. The bad news is that of the three applications I am interested in (simul-beacons, RawLogViewer, and rbpf-slam) the latter two crash with no error messages (i.e. "the application has stopped working" Windows pop up). Are you close to posting the 1.4 binaries for Windows?

jlblancoc, 2016-04-14:
Hi, 

If you tested it after the commit 23cb56f , then it looks like a problem with wxWidgets DLLs... make sure of building wxWidgets yourself so the mingw versions are compatible, use "dependcy walker" to make sure the .exe's find the correct DLLs, and if it still fails use Windows System Error Logs to identify the exact error cause...

prpsymbol, 2016-04-14:
Hi. I did build the wxWidgets myself with the same toolchain. I ran the dependency walker tool, but I have no idea what I am looking at. It seems to be referencing the correct dlls (along with a thousand Microsoft dlls). The zip file contains the .dwi file. Meanwhile, I'll go look in the Windows error logs.

[rbpf-slam.zip](https://github.com/MRPT/mrpt/files/219620/rbpf-slam.zip)

prpsymbol, 2016-04-14:
Here are the Windows Event Logs for the two crashes. rbpf-slam crashes when I run it, but rawlogviewer only crashes when I go to "raw map" and click "map from odometry".
[rawlogviewerEventLog.txt](https://github.com/MRPT/mrpt/files/219646/rawlogviewerEventLog.txt)
[rbpfslamEventLog.txt](https://github.com/MRPT/mrpt/files/219647/rbpfslamEventLog.txt)

prpsymbol, 2016-04-14:
I don't know what I grabbed for the rawlogviewer error, but here is the error similar to rbpf-slam:
[rawlogviewerEventLog2.txt](https://github.com/MRPT/mrpt/files/219651/rawlogviewerEventLog2.txt)

jlblancoc, 2016-04-14:
Thanks! From all your info, it seems that there are actually more MinGW-specific errors... 

I can't debug it myself soon, but if you are interested in trying, the next step would be to build, for example, `rbpf-slam` in Debug mode (if you use QtCreator, I think it is set under "Project" -> Build -> ...). You don't need to build the entire MRPT, just rbpf-slam. Also, add the `-j4` to the mingw-make arguments to speed it up. 
When built, hit "Debug" (F5) and let me know where does it find the crash problem. The most interesting part would be the call stack at the instant of the crash (a screenshot will be great!)

Sorry for the inconveniences this is causing you! We haven't tested MinGW in a long time, so it seems some errors have been accumulating for this compiler...

prpsymbol, 2016-04-14:
I haven't been using Qt for this so far, as I've been building using cmake/mingw32-make from the command line, but I can set it up to run under Qt (but I'll probably have to rebuild all of mrpt).

prpsymbol, 2016-04-15:
I've had an interesting 2 days, as my main development PC crashed yesterday, which forced me to bring up the mrpt dev environment on another machine. Of course, when I got through with all of the installs and builds, the apps of interest to me (simul-beacons, rawlogviewer, and rbpf-slam) now all work just fine. I am not 100% sure what the difference is, but I suspect I did not use the USE_OPENGL=1 flag in the original build of wxWidgets. The build I did today used the following command:

mingw32-make -f makefile.gcc CXXFLAGS="-std=gnu++11" SHARED=1 USE_OPENGL=1 DEBUG_INFO=0 VENDOR=mrpt UNICODE=1 BUILD=release

Once my IT Support guys revive my main PC, I see if I can verify that was the difference.

jlblancoc, 2016-04-15:
Good! :+1:

commit 23cb56f1 (2016-04-13): Fix serious mem bug in MinGW

Closes #225
closed 2016-04-15 by jlblancoc