MRPT/mrpt #229: mrpt FTBFS on arm64

jlblancoc, 2016-04-18:
mrpt FTBFS on arm64:
```
| [ 54%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/registerAllClasses.cpp.o
| cd /«PKGBUILDDIR»/obj-aarch64-linux-gnu/libs/maps && /usr/bin/c++   -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -Dmrpt_maps_EXPORTS -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2  -isystem /usr/include/eigen3/Eigen -isystem /usr/include/opencv -isystem /usr/include -pthread -I /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -I /usr/include/wx-3.0 -isystem /usr/include/wx-3.0 -Wreturn-type -Wextra -Wno-unused-parameter  -Wall -Wno-long-long -Wno-variadic-macros -Wno-write-strings -std=c++11 -pthread   -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/eigen3 -I/usr/include/eigen3/unsupported -I/usr/include/opencv -I/usr/include/aarch64-linux-gnu -I/usr/include/aarch64-linux-gnu/ffmpeg -I/usr/include/aarch64-linux-gnu/libavcodec -I/usr/include/aarch64-linux-gnu/libavformat -I/usr/include/aarch64-linux-gnu/libswscale -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/«PKGBUILDDIR»/. -I/«PKGBUILDDIR»/obj-aarch64-linux-gnu/include/mrpt-config/unix -I/«PKGBUILDDIR»/libs/maps/include -I/«PKGBUILDDIR»/libs/obs/include -I/«PKGBUILDDIR»/libs/opengl/include -I/«PKGBUILDDIR»/libs/base/include -I/«PKGBUILDDIR»/libs/maps/src    -o CMakeFiles/mrpt-maps.dir/src/registerAllClasses.cpp.o -c /«PKGBUILDDIR»/libs/maps/src/registerAllClasses.cpp
| [ 54%] Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/CFeatureExtraction_SURF.cpp.o
| cd /«PKGBUILDDIR»/obj-aarch64-linux-gnu/libs/vision && /usr/bin/c++   -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -Dmrpt_vision_EXPORTS -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2  -isystem /usr/include/eigen3/Eigen -isystem /usr/include/opencv -isystem /usr/include -pthread -I /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -I /usr/include/wx-3.0 -isystem /usr/include/wx-3.0 -Wreturn-type -Wextra -Wno-unused-parameter  -Wall -Wno-long-long -Wno-variadic-macros -Wno-write-strings -std=c++11 -pthread   -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/eigen3 -I/usr/include/eigen3/unsupported -I/usr/include/opencv -I/usr/include/aarch64-linux-gnu -I/usr/include/aarch64-linux-gnu/ffmpeg -I/usr/include/aarch64-linux-gnu/libavcodec -I/usr/include/aarch64-linux-gnu/libavformat -I/usr/include/aarch64-linux-gnu/libswscale -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/«PKGBUILDDIR»/. -I/«PKGBUILDDIR»/obj-aarch64-linux-gnu/include/mrpt-config/unix -I/«PKGBUILDDIR»/libs/vision/include -I/«PKGBUILDDIR»/libs/obs/include -I/«PKGBUILDDIR»/libs/opengl/include -I/«PKGBUILDDIR»/libs/base/include -I/«PKGBUILDDIR»/libs/vision/src    -o CMakeFiles/mrpt-vision.dir/src/CFeatureExtraction_SURF.cpp.o -c /«PKGBUILDDIR»/libs/vision/src/CFeatureExtraction_SURF.cpp
| In file included from /usr/include/eigen3/unsupported/Eigen/MatrixFunctions:56:0,
|                  from /«PKGBUILDDIR»/libs/vision/include/mrpt/vision/CDifodo.h:17,
|                  from /«PKGBUILDDIR»/libs/vision/include/mrpt/vision.h:35,
|                  from /«PKGBUILDDIR»/libs/vision/src/registerAllClasses.cpp:13:
| /usr/include/eigen3/unsupported/Eigen/src/MatrixFunctions/MatrixExponential.h: In function 'void Eigen::internal::matrix_exp_compute(const MatrixType&, ResultType&)':
| /usr/include/eigen3/unsupported/Eigen/src/MatrixFunctions/MatrixExponential.h:351:33: error: 'StdStemFunctions' was not declared in this scope
|      result = arg.matrixFunction(StdStemFunctions<ComplexScalar>::exp);
|                                  ^
| /usr/include/eigen3/unsupported/Eigen/src/MatrixFunctions/MatrixExponential.h:351:63: error: expected primary-expression before '>' token
|      result = arg.matrixFunction(StdStemFunctions<ComplexScalar>::exp);
|                                                                ^
| libs/vision/CMakeFiles/mrpt-vision.dir/build.make:761: recipe for target 'libs/vision/CMakeFiles/mrpt-vision.dir/src/registerAllClasses.cpp.o' failed
| make[4]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/src/registerAllClasses.cpp.o] Error 1

```
Please check
https://buildd.debian.org/status/fetch.php?pkg=mrpt&arch=arm64&ver=1:1.3.2-1&stamp=1449689457
for a full build log.

MarianoJT88, 2016-04-20:
Please check that it has been fixed!

jlblancoc, 2016-04-20:
:+1: for me if you tested it and it works with a real RGBD dataset! ;-) 

Just one more thing: Please check if this line can be removed now: 
https://github.com/MRPT/mrpt/blob/9f72d70abe8db107f7331f3723cf2553d7c307d0/libs/vision/include/mrpt/vision/CDifodo.h#L17

MarianoJT88, 2016-04-20:
Yes I have tested it with both datasets and the real camera, but I cannot now if it fixes the original problem...

I cannot remove that line because I am still using the log matrix function :S

jlblancoc, 2016-04-20:
Here you have both exp() and log() ;-)
http://reference.mrpt.org/devel/structmrpt_1_1poses_1_1_s_e__traits_3_013_01_4.html

MarianoJT88, 2016-04-21:
Done! 
I prefer the exact methods :) , they don't slow down the algorithm because they are called few times per frame.

jlblancoc, 2016-04-21:
Perfect!! 

Muchas gracias :-)

closed 2016-04-21 by jlblancoc

MarianoJT88, 2016-04-21:
De nada!

Date: Thu, 21 Apr 2016 03:49:24 -0700