MRPT/mrpt #234: Segmentation Fault under MinGW/Win 8.1

prpsymbol, 2016-04-21:
Hi. I am still having issues with running rbpf-Slam. I previously reported a successful build and working applications on an alternate PC running WinXP, closing Issue #225. I have now duplicated the entire toolchain on my main PC running Win8.1, and I still have the same problem.

rbpf-slam is getting a Segmentation Fault (signal SIGSEGV) when executing a call to copyPacket in the DenseCoeffsBase.h file (line 537), after calling mostLikMap->saveMetricMapRepresentationToFile from the rbpf-slam.cpp file (line 414), at least that's where the debugger stops. There is also another thread that is stopped at a call to aligned_malloc in memory.h at line 224.

Any idea what may be going on here?

prpsymbol, 2016-04-21:
When running RawLogViewer, it also crashes when I hit the "map from odometry" button, here is the error output I get:

Debugging starts
SHIMVIEW: ShimInfo(Complete)
Unrecognized accel key 'create stereo images observations from pairs of monocular images', accel string ignored.
Unrecognized accel key 'create stereo images observations from pairs of monocular images', accel string ignored.
In file ..\..\include/wx/msw/private.h at line 800: 'UnregisterClass' failed with error 0x00000584 (class still has open windows.).
Assertion failed!

Program: C:\Users\e815228\Documents\MRPT\MRPTQTBuildDebug\bin\RawLogViewer.exe
File: C:/Users/e815228/Documents/MRPT/MRPT-SVN/otherlibs/eigen3/Eigen/src/Core/DenseStorage.h, Line 86

Expression: (reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"
Debugging has finished

jlblancoc, 2016-04-21:
It may be not a solution, but a workaround, but... it seems to be that you are building in 32bit, aren't you? Try recompiling everything in 64bit and, at least, the Eigen errors should, in theory, go away. 

It is worth researching anyway why it works with 32bit builds in Visual Studio...

prpsymbol, 2016-04-21:
Tried to compile/link with TDM-GCC-64 but make dies at Linking CXX shared library ..\..\bin\libmrpt-base140-dbg.dll with a ton of unresolved references to gz* functions.

jlblancoc, 2016-04-21:
Try after this small change to see it solves the gz* issue...
51506d574a90c5a75f024ea012e21245bc960c91

prpsymbol, 2016-04-22:
Still broken. I'm giving up and switching to Visual Studio. I have installed the 1.4.0 binaries and I do not want to build mrpt, just want to get going with a first program. How do I set up the environment in Visual Studio to do that?

jlblancoc, 2016-04-22:
Ok... Take a look at this first guide, hope it's good to start: http://www.mrpt.org/Writing_applications_with_the_MRPT_C%2B%2B_Library

closed 2016-04-22 by jlblancoc