MRPT/mrpt #238: COpenNI2Generic::CDevice::open should not require COLOR_STREAM

jolting, 2016-04-26:
Right now COpenNI2Generic::CDevice::open() is making an assumption about both the COLOR_STREAM and the DEPTH_STREAM are always being available.

Maybe it is safe to assume the depth sensor is always available, but in the Occipital Structure sensor the color sensor is never available.

Currently, if the color sensor is unavailable then the depth sensor is never opened.
If the depth sensor is not available then the color sensor is never started.

https://github.com/MRPT/mrpt/blob/master/libs/hwdrivers/src/COpenNI2Generic.cpp#L727
```c++
	for(int i = 0;i < STREAM_TYPE_SIZE;++i){
		if(m_streams[i]->open(w, h, fps) == false){
			return false;
		}
	}
```

COpenNI2Generic::CDevice::open() should probably attempt to open any stream that is currently available and enable registration if both depth and color is available.

jlblancoc, 2016-04-26:
You're right... I committed yesterday without testing with a real Xtion sensor. Now I can confirm the problem, in my case because the IR channel can't be opened. 
Will try to fix it...

BTW: I don't understand the change in your (Steve's) code from `IMAGE_REGISTRATION_DEPTH_TO_COLOR` to `IMAGE_REGISTRATION_OFF` in `setImageRegistrationMode()`... any comment to this?

jlblancoc, 2016-04-26:
@jolting I think it's done now... Tested with an Asus Xtion, it tried to open all 3 channels but doesn't crash if this guy only emits RGB and Depth (no IR). 

However, take a look just in case you may need to add some method to retrieve the IR image which, I think is now not written anywhere (!?): only RGB & depth are accessed [here](https://github.com/MRPT/mrpt/blob/master/libs/hwdrivers/src/COpenNI2Generic.cpp#L845).

closed 2016-04-26 by jlblancoc

jolting, 2016-04-26:
> BTW:
> I don't understand the change in your (Steve's) code from IMAGE_REGISTRATION_DEPTH_TO_COLOR to IMAGE_REGISTRATION_OFF in setImageRegistrationMode()... any comment to this?

I'll have to take a look. IR and Depth streams from the same sensor, so it's unnecessary to register these streams. My assumption is since IR and DEPTH are the only two available maybe there is something that requires registration to be disabled, but I'm not certain.

>However, take a look just in case you may need to add some method to retrieve the IR image which, I think is now not written anywhere (!?): only RGB & depth are accessed here.

The 2D IR is useful for understanding issues with the depth sensor, but merging it with a 3D observation is probably not useful.

I was thinking of providing a function with the identical type signature to the getNextFrameRGB.
```c++
bool COpenNI2Generic::CDevice::getNextFrameIR(mrpt::utils::CImage &img, uint64_t &timestamp, bool &there_is_obs, bool &hardware_error);
```

commit db4a2e43 (2016-04-26): more robust COpenNI2Generic

Closes #238