MRPT/mrpt #240: simul-beacon and rbpf-slam accuracy questions

prpsymbol, 2016-04-29:
Hi JL,

I am working with simul-beacons and I want to make sure I am interpreting the
results I am getting correctly. I have modified simul-beacons to output a 
ground-truth path (from realPose) to compare to the estimated path, and
simul-beacons already outputs the ground-truth beacon map (outSimMap.txt).
I am using 16 beacons randomly placed in a 20x20m space, and the robot path is
either a centered 10m square or a centered 10m diamter circle, and I am
simulating a path that goes around the perimeter a little more than twice.

Output from rbpf-slam:
- log_estimated_path.txt : I am assuming this is the robot path estimated from
  the observations, correct? Essentially the main localization output.
- mapbuilt_xxxx_CBeaconMap_00_covs.txt: I am assuming that the last file in
  this xxxxx series represents the algorithm's final view of where the beacons
  are.
- log_odometry_path.txt : this appears to be exactly what the noisy odometry is
  delivering to the algorithm.
- Is there a viewer for .simplemap files?
- most_likely_path.txt : what is this?

So after the run, I am comparing the g-t path and the log_estimated_path.txt
file, and the g-t beacon and the mapbuilt data. Is this correct?

If I keep the odometry and distance sensor errors small, everything tracks
very nicely. However, if I keep the odometry error to 0.020 but bump up the
sensor error to 1.0, I am getting some results I do not fully understand.

The beacon map errors seem very large and all over the place:

[BeaconMapStats] // the data is based on the distance between the est & g-t
Min=0.259220
Max=14.006935
Mean=5.933765
Var=20.959036
Sig=4.578104

The robot path data is also a little odd to me:

[RobotPathStatsFirstHalf]
Min=0.032695
Max=2.627081
Mean=1.731305
Var=0.551524
Sig=0.742647

[RobotPathStatsSecondHalf]
Min=1.395121
Max=2.890823
Mean=2.268059
Var=0.164650
Sig=0.405770

Small variance, but a fairly large mean. Am I looking at this correctly?
Is the large sensor sigma driving this beyond the algorithm's capability?
Interestingly, the circular path yields better results on the robot path stats
(smaller mean).

I would appreciate any insights you can give me!
-Paul

jlblancoc, 2016-05-02:
Hi!

> I am working with simul-beacons and I want to make sure I am interpreting
> the
> results I am getting correctly. I have modified simul-beacons to output a
> ground-truth path (from realPose) to compare to the estimated path, and
> simul-beacons already outputs the ground-truth beacon map (outSimMap.txt).

Good! If you think it may be useful, feel free to open a pull-request ;-)

> Output from rbpf-slam:
> log_estimated_path.txt : I am assuming this is the robot path estimated from
> the observations, correct? Essentially the main localization output.

Right. More precisely, it is the average or mean of the robot path
hypotheses (see "most likely" below...)

> mapbuilt_xxxx_CBeaconMap_00_covs.txt: I am assuming that the last file in
> this xxxxx series represents the algorithm's final view of where the beacons
> are.

Well, I can't remember right now but you should get `mean` and a `cov`
file pairs. Then, yes, it's the evolution of the map estimation.

> log_odometry_path.txt : this appears to be exactly what the noisy odometry
> is delivering to the algorithm.

Sure.

> Is there a viewer for .simplemap files?

Unfortunately, there is not. However, read more on this here:
https://github.com/MRPT/mrpt/wiki/Ideas-page-for-MRPT-Google-Summer-of-Code-2016#project-6-new-gui-app-to-visualize-and-edit-robot-made-maps

That Windows viewer should only capable of view objects whose
serialization number have not changed in 8-9 years, so it may be near
to useless anyway...

You can also see the 3D visualization of maps in another rbpf-slam log
directory.

> most_likely_path.txt : what is this?

The path of the RBPF particle with the highest weight. Should be close
to the mean, but not exactly the same thing.

> So after the run, I am comparing the g-t path and the log_estimated_path.txt
> file, and the g-t beacon and the mapbuilt data. Is this correct?

Yes.

> If I keep the odometry and distance sensor errors small, everything tracks
> very nicely. However, if I keep the odometry error to 0.020 but bump up the
> sensor error to 1.0, I am getting some results I do not fully understand.

To further analyze the problem I would need to know the exact RBPF
algorithm (map representation): sum of gaussians or MonteCarlo?
Also, the exact spatial disposition of path & beacons may be leading
here to some degenerate solutions... that would be my wild guess.

Anyway, you could share the RBPF-SLAM .ini file you use and a sample
.rawlog to help reproducing the error.

Cheers,
JL

prpsymbol, 2016-05-02:
Hi JL,

I'm not very familiar with working in github, so I'll hold off on the pull request for now. I am attaching two zip files. The first is specific to the issue I posted, with getting very large beacon errors, but rather small position estimation errors. The second contains some of the things I was looking at today, which I call the wandering beacon problem. 

BTW, I modified simul-beacons (locally) to create beacon layouts on a grid. Makes it easier to see the behavior of the SLAM algorithm when you know where the beacons are supposed to be.

I really appreciate you taking the time to look this over!

-Paul
[BigBErrSmallPErr.zip](https://github.com/MRPT/mrpt/files/245945/BigBErrSmallPErr.zip)
[WanderingBeacon.zip](https://github.com/MRPT/mrpt/files/245946/WanderingBeacon.zip)

jlblancoc, 2016-05-11:
Hi, 

I just replicated your problem. 
It seems to be caused by the estimator becoming overconfident due to an "assumed" range error standard deviation (0.02 m) smaller than the real one used to simulate the dataset (apparently, 0.5 m). 

Changing the RBPF config file to something like the following makes the map to converge to a nice rectangular grid of beacons ;-) 

```
# Likelihood Options for BeaconMap 00:
[MappingApplication_beaconMap_00_likelihoodOpts]
rangeStd=0.2

```

closed 2016-05-11 by jlblancoc