MRPT/mrpt #243: Comparison of double and floating point in collision grid meta data

jolting, 2016-05-02:
Is it safe to add some epsilon in here, so that collision grids don't regenerate if they are close enough?

https://github.com/MRPT/mrpt/blob/master/libs/nav/src/tpspace/CParameterizedTrajectoryGenerator.cpp#L625

jolting, 2016-05-03:
@jlblancoc Actually this bug is more severe than I originally thought.
Collision grids will always regenerate.

This commit broke it: 
https://github.com/MRPT/mrpt/commit/e5db1da58bcc75422728efbdff6e623bc24c33a0

This writes doubles to the file.
```c++
		*f << m_x_min << m_x_max << m_y_min << m_y_max;
		*f << m_resolution;
```
And this reads floats.
```c++
		*f >> ff; if(ff!=m_x_min) return false;
		*f >> ff; if(ff!=m_x_max) return false;
		*f >> ff; if(ff!=m_y_min) return false;
		*f >> ff; if(ff!=m_y_max) return false;
		*f >> ff; if(ff!=m_resolution) return false;
```

It doesn't appear the MRPT does any type checking when deserializing.

jlblancoc, 2016-05-03:
Thanks for the discovery @jolting !!

I'll verify it and correct it ASAP.
Type-checking if only done at the object (`CSerializable` and derived classes) level. POD are not checked to avoid the huge overload that it would represent...

jlblancoc, 2016-05-03:
@jolting plase, get the changes from that commit above.

commit c6e6ece7 (2016-05-03): Fix PTG LUT will always regenerate (Closes #243 )
closed 2016-05-03 by jlblancoc