MRPT/mrpt #246: _glutInit on OS X 10.11.4 link error.

autosquid, 2016-05-03:
Here is the link error I met when building on max.

> Undefined symbols for architecture x86_64:
>   "_glutInit", referenced from:
>       mrpt::gui::CMyGLCanvasBase::Render() in CMyGLCanvasBase.o
>       mrpt::gui::CMyGLCanvasBase::InitGL() in CMyGLCanvasBase.o

clang so complains deprecated glutinit since OS X 10.9.

Am I missing extra libraries to install ?

jolting, 2016-05-11:
@autosquid 
Have you tried to install freeglut yet?
I don't have experience with this. I believe the version that Apple ships is a proprietary library.

SDL might be a better alternative to GLUT, but that code will have to be rewritten. Qt Has their own OpenGL wrapper which should be more portable.

jlblancoc, 2017-01-26:
One of the GSoC2017 proposed topics is such a port of wxWidgets-based `mrpt::gui` classes to Qt, in fact... 

@autosquid building in Travis for OSX with clang seems to be working fine, and since I don't have access to a Mac for testing, I can't further try to reproduce the error...  did you solve the problem? 

In principle, I'm closing the ticket, unless you can provide more details on how your build differs wrt Travis, for example, so there is some starting point to work on...

Thanks.

closed 2017-01-26 by jlblancoc