MRPT/mrpt #249: CPointsMap_liblas now uses boost uuids

jolting, 2016-05-12:
https://github.com/libLAS/libLAS/commit/953ca56240086d3be7324f3974693da4db4b52d4

https://github.com/MRPT/mrpt/blob/master/libs/maps/src/maps/CPointsMap_liblas.cpp#L117

Maybe something like:
```c++
#if LIBLAS_VERSION_NUM < 1800
	out_headerInfo.project_guid       = header.GetProjectId().to_string();
#else
	out_headerInfo.project_guid       = boost::uuids::to_string(header.GetProjectId());
#endif
```

jlblancoc, 2016-05-12:
Ok, thanks for the pointer!

That would imply depending on boost headers, but it's correctly set as a dependency of [liblas-dev](http://packages.ubuntu.com/wily/liblas-dev), which is optional for MRPT, so let's do it as you recommend...

jlblancoc, 2016-05-12:
It's fixed, but only tested with libLAS 1.7.0. 
Will launch a rebuild in PPA to check it more thoroughly...

closed 2016-05-12 by jlblancoc

jolting, 2016-05-12:
@jlblancoc It builds now for 14.04 15.10, 16.04 and 16.10.

jlblancoc, 2016-05-12:
:+1: Great! Thanks for the note 

PS: I couldn't actually check it in my PPA because liblas is not marked as a dependency to save >100MB of disk space in /usr space that takes this library plus all its dependencies. But I think it's okay like that for official releases due to the really few features that MRPT offers related to LAS file formats...

jolting, 2016-05-12:
Yep. Since the builds happen on launchpad, it saves a lot of space. Although build dependencies aren't really a problem, if liblas has a lot of binary dependencies maybe the mrpt code which depends on liblas could be moved to another library. I'm thinking libmrpt-las. That way the debian packaged version is as feature complete as possible, but the base install has few dependencies.

commit c1bba637 (2016-05-12): Fixed #249