MRPT/mrpt #253: RBPF Getting the localization between localizations

jolting, 2016-05-13:
What's the best way to get a good pose estimation between localizations when using the RBPF map builder?

I was thinking of adding a get method to the protected member `CMetricMapBuilderRBPF::CMetricMapBuilderRBPF::odoIncrementSinceLastLocalization` and adding it to the pose I get from `CMetricMapBuilderRBPF::getCurrentPoseEstimation()`

jlblancoc, 2016-05-13:
Yes, that would be a good idea. Feel free of doing it and PR. 

Also, consider the use (in your user code, or inside a new method into RBPF map builder) of [CRobot2DPoseEstimator](https://github.com/MRPT/mrpt/blob/master/libs/base/include/mrpt/poses/CRobot2DPoseEstimator.h), which already implements something almost identical to what you propose...

jlblancoc, 2016-10-03:
Hi @jolting , this issue should be closed after https://github.com/MRPT/mrpt/commit/e90941ccbed5e70b0017ebc4a05950683e726f1a and the other new class in our repo. 

Cheers.

closed 2016-10-03 by jlblancoc