MRPT/mrpt #312: Integrate with ROS - Add Support for online execution

bergercookie, 2016-08-23:


bergercookie, 2016-09-27:
@jlblancoc I've implemented a inter-process communication strategy over TCP Sockets for transmitting measurements fetched from ROS topics (using a ROS node) over to the CGraphSlamEngine application. This enables the graphslam-engine application to be executed in an online fashion in parallel with the measurement acquisition process.

Could you open a new gh topic to which I can PR my changes for further testing and discussion on this?

The branch at hand is [graphslam-devel](https://github.com/bergercookie/mrpt/commits/graphslam-devel)

jlblancoc, 2016-09-27:
Great news, @bergercookie !! 

I wonder whether you chose this approach instead of wrapping the existing app as a ROS node, avoiding the need to serialize / deserialize in a custom way... but if it's done and it works, it's welcome! ;-)

I created the [new branch](https://github.com/MRPT/mrpt/tree/graphslam-devel) for you to open a PR, but **please** wait until [travis ends](https://travis-ci.org/MRPT/mrpt/builds/163089814) because we're still solving some errors with wx2.8 after another PR earlier today, and I would like yours to start from a clean build...  If travis fails, I'll "rebase" the branch to a future commit when all is ok.

bergercookie, 2016-09-27:
>I wonder whether you chose this approach instead of wrapping the existing app as a ROS node, avoiding the need to serialize / deserialize in a custom way... but if it's done and it works, it's welcome! ;-)

Indeed, this is a cleaner apprach. However given the current implementation of the CGraphSlamEngine class it is not really feasible as this class contains a parseRawlogFile method that does all the heavy work. 

>I created the new branch for you to open a PR, but please wait until travis ends because we're still solving some errors with wx2.8 after another PR earlier today, and I would like yours to start from a clean build... If travis fails, I'll "rebase" the branch to a future commit when all is ok. 

Ok, I will make the PR after the CI is executed successfully.

Thanks,

jlblancoc, 2016-09-27:
Travis is green , go on ! https://travis-ci.org/MRPT/mrpt/builds/163089814

jlblancoc, 2016-09-27:
Closed after #333

closed 2016-09-27 by jlblancoc