MRPT/mrpt #313: graphslam-engine: Add support for multi-robot graphSLAM

bergercookie, 2016-08-23:
@jlblancoc, @EduFdez
Goal: Be able to execute graphslam-engine in multi-robot setups. Final product should address the following issues:
- Implement node/edge decider classes that are able to incorporate measurements(?) from other robots. Aforementioned deciders should be used for intra-robot communication so that they can exchange measurements
- Concrete doxygen documentation on the procedure and the implemented node/edge registration deciders
- Application verifying that the end-result is working correctly
- Demo video(s) demonstrating the application capabilities

Notable works on the field:
- http://frc.ri.cmu.edu/~kaess/pub/Kim10icra.pdf
- http://webdiis.unizar.es/~mtlazaro/papers/Lazaro-IROS13.pdf

[Proposed Label] Enhancement
[Proposed Assignee] @bergercookie

jlblancoc, 2016-08-23:
:+1: 

Just a comment: when you work in your work, do it as you prefer, but when time comes for a pull-request, please open it against a new branch (not `master`) so all new code can be thoroughly tested in all platforms and compilers, etc.

bergercookie, 2016-08-23:
👍

bergercookie, 2016-12-01:
Hi @jlblancoc, @EduFdez

I have finished the real-time experiments using ROS and the algorithm that we developed during GSoC and now I am about to move on and design the mr-slam algorithm (using MRPT, obviously).

I am mostly concentrated on the strategy described in [this paper](http://webdiis.unizar.es/~mtlazaro/papers/Lazaro-IROS13.pdf) by M.T. Lazaro.

The main contribution of this paper is the communication model that minimizes data exchange between the agents so it is more suitable for online use.

--------------

To sum up the strategy:
- each robot executes, independently, single robot graph SLAM using the g2o backend.
- When in close range (robots use ping to find their neighbours) they start exchanging their local maps. Each robot sends a message that consists of the following:

    + latest measurement (laser scan) and the corresponding node ID
    + up to date estimated locations of the last N registered nodes (these positions change due to optimization so they are sent again in every message)

This message is sent every time a new node is added to the graph - to all the robots that are in range.

- Using the latter and having buffered the received laser scans, each receiver robot reconstructs the transmitter's local map. Using a RANSAC-based outlier rejection scheme and a correlative scan matching algorithm receiver robot builds a list of nodes that it has matched against its own local map.

- Receiver robots send back the aforementioned list of nodes and using this, transmitter sends out a condensed graph that contains these nodes and the edges between them (based on its own graph).
- Receivers finally incorporate the condensed graph in their own, and continue executing SLAM

---------

From my standpoint roughly, I should concentrate on implementing the following:
- Some network utilities in mrpt::utils::net namespace for agents (e.g. ping)
- New graphslam frontend classes (NRD, ERD) that enable batch node/edge acquisition and registration. Classes should take care of the robots communication as well.
- Methods in the mrpt::graphs::CNetworkOfPoses classes for extracting part of the graph (so that it can be sent to the other robot)
- Reformat the CLoopCloserERD class so that Olson's scheme is used for finding the candidate edges between the measurements of the robot's own nodes and those of the received nodes. This should be able to replace the RANSAC-based scheme used in the paper.

Am I missing something here? Is any of the described steps already implemented in MRPT?

With regards to testing this, I am currently building a simulation environment in Gazebo for doing it, but do you think of any way of testing the approach in MRPT (without any third-party tool)

Just wanted to give you a heads up and discuss the proposed strategy. :-)

Hope you are well,
Nikos

jlblancoc, 2016-12-03:
All makes sense, thanks for sharing the detailed working plan here!

Nope, the missing parts that you need are not already done in MRPT, they should be modified or added as you described above. 

Just a doubt: right now you are not relying on g2o, so that will be another difference in the implementation of GraphSLAMEngine?? I'm just worrying about the addition of new dependencies to MRPT, but if it is a must, we should do it carefully so compilation doesn't break if those new dependencies are not present in a system.

Cheers.

bergercookie, 2016-12-03:
> Just a doubt: right now you are not relying on g2o, so that will be another difference in the implementation of GraphSLAMEngine?? I'm just worrying about the addition of new dependencies to MRPT, [...]

Hmm, I don't think there is a fundamental difference in the algorithm if I use the standard LM solver instead of the g2o framework. I 'll start working on the networking/communications part and see if g2o is mandatory

jlblancoc, 2016-12-06:
BTW: Testing multirobot SLAM seems a perfect fit for our [msvim](https://github.com/ual-arm-ros-pkg/mvsim) environment ;-)

bergercookie, 2017-03-01:
Hey @jlblancoc, 

I am currently working on implementing a multi-robot graphSLAM problem following the method that I described in our previous conversation. My initial strategy was to have each robot agent use Olson's plan to find correspondences between own and other agents' LaserScans and using those, to transfer and integrate certain parts of other agents' trajectories and maps into own map. However, this method doesn't seem to be very robust and is very sensitive with regards to the initial estimation of the ICP algorithm that is used. 

I am currently investigating your map-merging technique that is described [here](http://www.mrpt.org/paper-occupancy-grid-matching/) as an alternative for finding a transformation between maps of running agents. I will be using this technique in the end of the SLAM operation anyway, so that i can have a final estimate of the whole map (by joining all the local robot maps).
- I am  using the CGridAligner class for finding a transformation between the two maps but I can't seem to find a method/way of actually rotating a COccupancyGridMap2D by a certain transformation and afterwards join this COccupancyGridMap with another and produce a final COccupancyGridMap2D object. Any suggestion on how to do this? Is there any code sample available?
- Also, the CGridAligner::AlignPDF method doesn't always compute the correct transformation between the two maps. Would you suggest of any way to speculate whether the proposed transformation is correct (e.g. a goodness value) ? Is there any parameter that I can tweak so that I can compute a more conservative, but correct transformation?

Thanks in advance,
Nikos

bergercookie, 2017-03-17:
Hi @jlblancoc,

FYI, I made a first working version of the multi-robot graphSLAM application.

The main points of the algorithm are outlined in [this report](https://github.com/MRPT/mrpt/files/850659/mr_graphslam_simulation.pdf) and a demo video is also available [here](https://www.youtube.com/watch?v=4RKS2jrvsYE&t=1s). The main difference with regards to the previously proposed plan is the use of map-matching module of MRPT instead of Olson's Loop closing scheme. This way the intra-graph transformation between each robot's graph and its neighbor's graph is computed and using that, the robot is able to integrate all neighbor's  fetched nodes and corresponding measurements.

Until now, I have tested the code in simulations with 2 robots in Gazebo and it seems to be working properly. I am planning on extending the testing on more robots as well as real experiments. The only sensitive part is the computation of the intra-graph transformation which requires the agents to have mapped a large portion of their environment first.

You can find more on the implementation in the following repositories:
- https://github.com/bergercookie/csl_mr_slam
- https://github.com/bergercookie/mrpt_slam/tree/graphslam-devel

When this is ready I will make the corresponding pull-requests.

jlblancoc, 2017-03-18:
wow, congrats for the great results! 

I'm happy the gridmap aligning method worked for this app.

Best luck towards a successful thesis writing... that's one of the toughest parts, but you'll be forced to think a lot about all the subsystems as you write and that is always good to have a bird-eye view of the problem ;-)

bergercookie, 2017-06-12:
Closing this via cc0cdd17377d9b7cd0bae138e4a69c434331667f

closed 2017-06-12 by bergercookie