MRPT/mrpt #315: graphslam-engine: Node Reduction scheme

bergercookie, 2016-08-23:
**Goal:** Put a upper bound on the maximum number of nodes that can be registered in a graph. 

_Note:_ This should be an invaluable feature in datasets with long robot trajectories since the computational complexity of the problem (maintaining the necessary structs for constraints, full optimization of the graph) increase with the number of registered nodes 

Notable works in the field:
- http://ais.informatik.uni-freiburg.de/teaching/ws12/mapping/pdf/slam16-hierarchical-4.pdf
- http://www2.informatik.uni-freiburg.de/~stachnis/pdf/kretzschmar12ijrr.pdf

[Proposed Label] Enhancement

jlblancoc, 2016-08-23:
How does this topic relate with "hierarchical maps", like:
* "HOG-Man" *G. Grisetti, R. Kummerle, C. Stachniss, U. Frese, and C. Hertzberg, “Hierarchical optimization on manifolds for online 2d and 3d mapping,” in IEEE Intl. Conf. on Robotics and Automation (ICRA), 2010*
* Submapping based on min N-cut, etc. (you already know some papers by myself).

??

bergercookie, 2016-08-23:
>HOG-Man" G. Grisetti, R. Kummerle, C. Stachniss, U. Frese, and C. Hertzberg, “Hierarchical optimization on manifolds for online 2d and 3d mapping,” in IEEE Intl. Conf. on Robotics and Automation (ICRA), 2010

As a matter of fact I accidentally put a link to Stachniss' Mapping course as I wanted to put a link to the HOG-Man strategy instead!

>Submapping based on min N-cut, etc. (you already know some papers by myself).
Indeed! I hadn't really thought of it before but partitioning and merging some of the nodes when they add up to a certain number based on your partitioning strategy is a viable option as well!

jlblancoc, 2022-09-21:
I'm afraid this one is outdated by now... :-( 
Closing it...

closed 2022-09-21 by jlblancoc