MRPT/mrpt #323: 2060907 mrpt origin/master make error

9578577, 2016-09-07:
/home/linux/mrpt/python/src/pnp_bindings.cpp:4:32: fatal error: numpy/arrayobject.h: No such file or directory
compilation terminated.
python/CMakeFiles/pymrpt.dir/build.make:326: recipe for target 'python/CMakeFiles/pymrpt.dir/src/pnp_bindings.cpp.o' failed
make[2]: *** [python/CMakeFiles/pymrpt.dir/src/pnp_bindings.cpp.o] Error 1
CMakeFiles/Makefile2:7310: recipe for target 'python/CMakeFiles/pymrpt.dir/all' failed
make[1]: *** [python/CMakeFiles/pymrpt.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

jolting, 2016-09-07:
Please post more info about what you're trying to do. 

The python bindings require numpy now.

If on ubuntu:
sudo apt-get install python-numpy

Alternatively, install numpy with pip.

jlblancoc, 2016-09-20:
Thanks for the note, @jolting !
I've updated the [wiki instructions](https://github.com/MRPT/mrpt/wiki/PythonBindings) on Python.

closed 2016-09-20 by jlblancoc