MRPT/mrpt #325: MRPT compilation on ARM architecture - rpnp.h:45:42: error: Eigen::Ref has not been declared

bergercookie, 2016-09-19:
Compilation fails on MRPT when compiling in Ubuntu 12.04 on ARM architecture.
@mangi020 could you assist on fixing this one?

System Information is provided [here](https://github.com/MRPT/mrpt/files/480223/sys_info.txt)

Full output of error is provided below:
-------------------------------------
[ 79%] [ 79%] [ 79%] [ 79%] Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/rpnp.cpp.o
Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/p3p.cpp.o
Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/lhm.cpp.o
Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/dls.cpp.o
In file included from /home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:21:0:
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.h:45:42: error: ‘Eigen::Ref’ has not been declared
                 bool compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_);
                                          ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.h:45:45: error: expected ‘,’ or ‘...’ before ‘<’ token
                 bool compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_);
                                             ^
In file included from /home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:18:0:
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.h:74:42: error: ‘Eigen::Ref’ has not been declared
                 bool compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_);
                                          ^
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.h:74:45: error: expected ‘,’ or ‘...’ before ‘<’ token
                 bool compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_);
                                             ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:51: error: ‘bool mrpt::vision::pnp::rpnp::compute_pose’ is not a static member of ‘class mrpt::vision::pnp::rpnp’
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                   ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:44: error: ‘Ref’ is not a member of ‘Eigen’
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                            ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:70: error: expected primary-expression before ‘>’ token
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                                      ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:72: error: ‘R_’ was not declared in this scope
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                                        ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:76: error: ‘Ref’ is not a member of ‘Eigen’
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                                            ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:102: error: expected primary-expression before ‘>’ token
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                                                                      ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:104: error: ‘t_’ was not declared in this scope
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                                                                        ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:42:106: error: expression list treated as compound expression in initializer [-fpermissive]
 bool mrpt::vision::pnp::rpnp::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                                                                          ^
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:43:1: error: expected ‘,’ or ‘;’ before ‘{’ token
 {
 ^
make[3]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/rpnp.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:189:31: error: ‘bool mrpt::vision::pnp::lhm::compute_pose’ is not a static member of ‘class mrpt::vision::pnp::lhm’
 bool lhm::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                               ^
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:189:24: error: ‘Ref’ is not a member of ‘Eigen’
 bool lhm::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                        ^
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:189:50: error: expected primary-expression before ‘>’ token
 bool lhm::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                  ^
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:189:52: error: ‘R_’ was not declared in this scope
 bool lhm::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                    ^
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:189:56: error: ‘Ref’ is not a member of ‘Eigen’
 bool lhm::compute_pose(Eigen::Ref<Eigen::Matrix3d> R_, Eigen::Ref<Eigen::Vector3d> t_)
                                                        ^
/home/linaro/mrpt_src/libs/vision/src/pnp/lhm.cpp:189:82: error: expected primary-expression before ‘>’ token
[... 31 lines cut ...]
 bool mrpt::vision::pnp::p3p::solve(Eigen::Ref<Eigen::Matrix3d> R, Eigen::Ref<Eigen::Vector3d> t, Eigen::MatrixXd obj_pts, Eigen::MatrixXd img_pts)
                                                                   ^
/home/linaro/mrpt_src/libs/vision/src/pnp/p3p.cpp:80:93: error: expected primary-expression before ‘>’ token
 bool mrpt::vision::pnp::p3p::solve(Eigen::Ref<Eigen::Matrix3d> R, Eigen::Ref<Eigen::Vector3d> t, Eigen::MatrixXd obj_pts, Eigen::MatrixXd img_pts)
                                                                                             ^
/home/linaro/mrpt_src/libs/vision/src/pnp/p3p.cpp:80:95: error: ‘t’ was not declared in this scope
 bool mrpt::vision::pnp::p3p::solve(Eigen::Ref<Eigen::Matrix3d> R, Eigen::Ref<Eigen::Vector3d> t, Eigen::MatrixXd obj_pts, Eigen::MatrixXd img_pts)
                                                                                               ^
/home/linaro/mrpt_src/libs/vision/src/pnp/p3p.cpp:80:114: error: expected primary-expression before ‘obj_pts’
 bool mrpt::vision::pnp::p3p::solve(Eigen::Ref<Eigen::Matrix3d> R, Eigen::Ref<Eigen::Vector3d> t, Eigen::MatrixXd obj_pts, Eigen::MatrixXd img_pts)
                                                                                                                  ^
/home/linaro/mrpt_src/libs/vision/src/pnp/p3p.cpp:80:139: error: expected primary-expression before ‘img_pts’
 bool mrpt::vision::pnp::p3p::solve(Eigen::Ref<Eigen::Matrix3d> R, Eigen::Ref<Eigen::Vector3d> t, Eigen::MatrixXd obj_pts, Eigen::MatrixXd img_pts)
                                                                                                                                           ^
/home/linaro/mrpt_src/libs/vision/src/pnp/p3p.cpp:80:146: error: expression list treated as compound expression in initializer [-fpermissive]
 bool mrpt::vision::pnp::p3p::solve(Eigen::Ref<Eigen::Matrix3d> R, Eigen::Ref<Eigen::Vector3d> t, Eigen::MatrixXd obj_pts, Eigen::MatrixXd img_pts)
                                                                                                                                                  ^
/home/linaro/mrpt_src/libs/vision/src/pnp/p3p.cpp:81:1: error: expected ‘,’ or ‘;’ before ‘{’ token
 {
 ^
make[3]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/p3p.cpp.o] Error 1
make[3]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/lhm.cpp.o] Error 1
make[2]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/all] Error 2
make[1]: *** [apps/graphslam-engine/CMakeFiles/graphslam-engine.dir/rule] Error 2
make: *** [graphslam-engine] Error 2

jlblancoc, 2016-09-19:
Could you give it a try replacing the `T::Ref` instances by standard `T &` references??

mcprakash, 2016-09-20:
Hi @bergercookie 
 Could you let me know how you are cross-compiling it or the specific device you are using ? 

 I compiled and built MRPT with pnp in Raspberry Pi (Arm V6 ) architecture with OpenCV and OpenGL support. Hence, I would like to get try it on your specific device before changing any of the codes.

 @jlblancoc  The use of `T::Ref` is to link shared memory of python Boost variable with Eigen matrix. Hence, changing it would be tedious. 

 @bergercookie Looking forward to your response. I will make quick corrections to enable fast fixes. 

Thanks 

Chandra

mcprakash, 2016-09-20:
Also @bergercookie 

 All the statements in your code are missing Eigen::Ref\<Eigen::Matrix3d\>. They are written as Eigen::RefEigen::Matrix3d.

 You can look at line 45 in the library https://github.com/MRPT/mrpt/blob/master/libs/vision/src/pnp/rpnp.h . It may be that '\<' or '\>' is not being displayed in as you need to place '\\\<' in github text. Please let me know if that is indeed the case so that I can look at the ARM v7 architecture. 

[ 79%] [ 79%] [ 79%] [ 79%] Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/rpnp.cpp.o
Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/p3p.cpp.o
Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/lhm.cpp.o
Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/dls.cpp.o
In file included from /home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.cpp:21:0:
/home/linaro/mrpt_src/libs/vision/src/pnp/rpnp.h:45:42: error: ‘Eigen::Ref’ has not been declared
**bool compute_pose(Eigen::RefEigen::Matrix3d R_, Eigen::RefEigen::Vector3d t_);**

 Best 

Chandra

bergercookie, 2016-09-20:
>All the statements in your code are missing Eigen::Ref<Eigen::Matrix3d>. They are written as Eigen::RefEigen::Matrix3d.

@mangi020 This has to be an issue with how github the pasted output. Anyway, here's the full output as captured using _tee_.
[20160919_pnp_compilation_error.txt](https://github.com/MRPT/mrpt/files/481880/20160919_pnp_compilation_error.txt)

bergercookie, 2016-09-20:
This has to be an issue with the Eigen debian packages. On odroid, the installed debian package is:
[libeigen3-dev_3.0.5-1](https://pkgs.org/ubuntu-12.04/ubuntu-universe-i386/libeigen3-dev_3.0.5-1_i386.deb.html) which **does not contain a Ref.h file**
However on newer versions (e.g. [libeigen3-dev_3.2.0-8_all.deb](https://pkgs.org/ubuntu-14.04/ubuntu-universe-i386/libeigen3-dev_3.2.0-8_all.deb.html) this file seems to be included)

jlblancoc, 2016-09-20:
`Eigen::Ref<>` was [introduced](http://eigen.tuxfamily.org/index.php?title=ChangeLog) in Eigen 3.2.0... 

But Ubuntu precise did come with an older version ( http://packages.ubuntu.com/precise/libeigen3-dev  ) so in that platform we should enforce building with MRPT's embedded copy of Eigen, which right now is 3.2.5. 

If building manually, could you please confirm that errors go away by setting MRPT CMake variable `EIGEN_USE_EMBEDDED_VERSION` to `ON`??

jlblancoc, 2016-09-20:
Update: PPA packages already took into account that Eigen version was too old in precise and enforce the embedded version in that case: 
https://github.com/MRPT/mrpt/blob/master/scripts/prepare_ubuntu_pkgs_for_ppa.sh#L43

bergercookie, 2016-09-20:
@jlblancoc indeed. By using the embedded version the compilation of mrpt-vision library is successful. Do you think we should check for the eigen version in vision-related code by checking the EIGEN_WORLD_VERSION, EIGEN_MAJOR_VERSION preprocessor macros?

jlblancoc, 2016-09-20:
Sure. There must be some logic to test for a minimum Eigen version somewhere in cmakescritps/script_eigen.cmake or alike... please, take a look for it and can propose a PR with a change of the minimum version to 3.2.0 for the logic to switch to the embedded version. Thanks!

bergercookie, 2016-09-20:
Sure! @jlblancoc , @mangi020  thanks for the support on this.

jolting, 2016-09-20:
The 12.04 support cycle has concluded. Is Linaro still using this for some purpose?

jlblancoc, 2016-09-21:
@jolting https://wiki.ubuntu.com/Releases 12.04.5 is still supported, if I understand it correctly, isn't it??

I'm adding now a commit to switch to embedded Eigen version if system version is < 3.2, so we could close this ticket, ok?

closed 2016-09-21 by jlblancoc

mcprakash, 2016-09-22:
> Closed #325 <https://github.com/MRPT/mrpt/issues/325> via b057b63

Hi Jose
 I am so sorry I could not be of much help. This entire week went by
fighting bed bugs. I am currently working on the real time camera pose
estimation application. It should be done shortly.

Best

Chandra

jolting, 2016-09-22:
Oh right. They went to a 5 year cycle in 12.04 desktop. I guess it was always 5 years for server. It concludes in April. That's not a lot of time for new development. They take all the packages off their mirrors when they end of life a release, so it's not good to let those deadlines sneak up on you. I've had that happen on a regular release, not an LTS. It's not fun when you can't fetch packages.

commit b057b63b (2016-09-21): require eigen 3.2 or use embedded version. Fix #325