MRPT/mrpt #338: Insertion of observation into CPointMap not thread safe.

jolting, 2016-10-04:
https://github.com/MRPT/mrpt/blame/master/libs/maps/src/maps/CPointsMap_crtp_common.h#L143

Calling reserve on a vector like this often can cause the entire vector to get reallocated and moved. It's not guaranteed to simply grow the vector in-place. Therefore, this operation is not thread safe. Also, the non-trivial case of the reserve may also increase overhead in many instances.

We caught this one with -fsanitize=address. It went unnoticed for quite a while.

I think the intuitive thing is to treat observations as const when inserting them, but I haven't considered how much effort it will be to add the type qualifiers.

jlblancoc, 2016-10-04:
> We caught this one with -fsanitize=address. It went unnoticed for quite a while.

wow!!  What a discovery: a built-in valgrind, so cool...

> Calling reserve on a vector like this often can cause the entire vector to get reallocated and moved. It's not guaranteed to simply grow the vector in-place. Therefore, this operation is not thread safe. Also, the non-trivial case of the reserve may also increase overhead.

Absolutely agree... I can think of a number of alternative ways to fix this, will come back later with one of them. 

Thanks for the research.

jlblancoc, 2016-10-04:
Please, read https://github.com/MRPT/mrpt/pull/340 for a discussion of the applied changes...

PR #340 (2016-10-04): Fix scan2d access
A simple change in CObservation2DRangeScan (hiding internal vectors) and enforce that all vectors have a 4*N size. 

This allows avoiding ugly tricks like the one that caused #338. 

API change: backwards compatible only if read-only access is done to range scans. Write access requires using the new methods, see https://github.com/MRPT/mrpt/blob/master/libs/obs/include/mrpt/obs/CObservation2DRangeScan.h

(Fixes #338)
closed 2016-10-04 by jlblancoc