MRPT/mrpt #345: Compute visible between two location points

simplePig, 2016-10-06:
Now i am doing a work about the visibility of two arbitrary points in grid map.
I have loaded grid map file but did not know how to realize.
Which API is able to determine whether is visible(path) or invisible(obstacle) between two location points ?
I can not find any available function in API of COccupancyGridMap2D.h

jolting, 2016-10-06:
https://github.com/MRPT/mrpt/blob/master/libs/maps/src/maps/COccupancyGridMap2D_simulate.cpp

I don't think there's anything that does that specifically, but if you take a look at `OccupancyGridMap2D::simulateScanRay` I think that does 90% of what you want to do.

start_x, start_y is the first point
angle_direction is the angle to your other point.
max_range_meters your distance between the two points.
noiseStd = 0
angleNoiseStd = 0
threshold_free is your occupancy grid threshold.

Perhaps a little more context about what you are trying to do might help. Are you really trying to see what is visible or are you trying to see if a direct path is navigable? Robots aren't points, so some of the TPSpace code might be interesting to you.

Perhaps you can ask your question on gitter, so that others can chime in.
https://gitter.im/MRPT/mrpt

simplePig, 2016-10-07:
3Q！ i have found a API named computePath can do that work , but i solved the problem by myself.

jlblancoc, 2016-10-07:
Good!

closed 2016-10-07 by jlblancoc