MRPT/mrpt #347: How to show on Qt's window?

NoBugLi, 2016-10-07:
I learned that COpenGLScenePtr object is a member variable of CDisplayWindow3D class and CMyGLCanvasBase class so that the OPENGL object can be shown in the corresponding window.Now I am using Qt as GUI.Could someone tell me that how to show the OPENGL object in mrpt on Qt's window?Thank you very much!

jlblancoc, 2016-10-07:
Hi @NoBugLi , 

There is no official support (yet) for direct Qt rendering, but in principle it is possible... and can be done. 

The Master student Muhammad Omer Saeed was very kind in sharing with us a prototype example of how to integrate Qt GUI with MRPT OpenGL classes, as part of his proposal for GSoC 2016. Please, find it attached. It could be a good reference implementation while a more formal support for Qt is introduced in mrpt-gui...  If yourself develop a reusable Qt component, suitable for rendering MRPT objects, feel free of opening a Pull-Request to contribute it back ;-) 

Cheers,
JL

[mrpt_qt_sample_applications.zip](https://github.com/MRPT/mrpt/files/515955/mrpt_qt_sample_applications.zip)

closed 2016-10-07 by jlblancoc

NoBugLi, 2016-10-08:
Very Great! Thanks.This demo can help me sovle this problem!