MRPT/mrpt #358: PnP code: fix build against old OpenCV versions

jlblancoc, 2016-10-23:
FTBFS for OpenCV 2.3:

Example error build log (https://launchpadlibrarian.net/290569576/buildlog_ubuntu-precise-armhf.mrpt_1%3A1.5.0~snapshot20161023-2003-git-7c0a5f92-precise-1~ppa1~precise_BUILDING.txt.gz):

```
/<<PKGBUILDDIR>>/libs/vision/src/pnp/dls.cpp: In member function ‘void mrpt::vision::pnp::dls::run_kernel(const cv::Mat&)’:
/<<PKGBUILDDIR>>/libs/vision/src/pnp/dls.cpp:121:57: error: no matching function for call to ‘cv::Mat::copyTo(cv::Mat)’
/<<PKGBUILDDIR>>/libs/vision/src/pnp/dls.cpp:121:57: note: candidates are:
/usr/include/opencv2/core/core.hpp:1651:10: note: void cv::Mat::copyTo(cv::OutputArray) const
/usr/include/opencv2/core/core.hpp:1651:10: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘cv::OutputArray {aka const cv::_OutputArray&}’
/usr/include/opencv2/core/core.hpp:1653:10: note: void cv::Mat::copyTo(cv::OutputArray, cv::InputArray) const
/usr/include/opencv2/core/core.hpp:1653:10: note:   candidate expects 2 arguments, 1 provided
/<<PKGBUILDDIR>>/libs/vision/src/pnp/dls.cpp:146:54: error: no matching function for call to ‘cv::Mat::copyTo(cv::Mat)’
/<<PKGBUILDDIR>>/libs/vision/src/pnp/dls.cpp:146:54: note: candidates are:
/usr/include/opencv2/core/core.hpp:1651:10: note: void cv::Mat::copyTo(cv::OutputArray) const
/usr/include/opencv2/core/core.hpp:1651:10: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘cv::OutputArray {aka const cv::_OutputArray&}’
/usr/include/opencv2/core/core.hpp:1653:10: note: void cv::Mat::copyTo(cv::OutputArray, cv::InputArray) const
/usr/include/opencv2/core/core.hpp:1653:10: note:   candidate expects 2 arguments, 1 provided
make[4]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/src/pnp/dls.cpp.o] Error 1
```

(cc: @mangi020 )

mcprakash, 2016-10-24:
This is a known error in OpenCV 2.3 [link](http://stackoverflow.com/questions/8005627/opencv-2-3-copyto-function-not-working-as-it-was-in-2-2). Should be fixed in few hours.

closed 2016-10-25 by jlblancoc

jlblancoc, 2016-10-25:
For the records: 
Well, it wasn't straightforward... Until now all these patches have been required to "fix" building under Ubuntu 14.04, each time through a PPA build cycle:

* https://github.com/MRPT/mrpt/commit/91ae7389855a566ff03aa7ea9e9860f864571c65
* https://github.com/MRPT/mrpt/commit/afbb90ff646bd0d0e8da89251e743390ba875fd3
* https://github.com/MRPT/mrpt/commit/88de04ffeae841b07807bd6904e608c983c14774
* https://github.com/MRPT/mrpt/commit/c4ad9b414bf85c6ca72aebab77d7645b1ae0a490
* https://github.com/MRPT/mrpt/commit/477542541af43565c8646d07d507ffa9d305cb17

commit 47754254 (2016-10-25): Workaround a problem with old opencv 2.3 in trusty

Fixes #358
A proper solution may be searched for, but I can't reproduce the build
errors in my system with OpenCV 2.3...
commit c4ad9b41 (2016-10-25): heavier workaround broken opencv 2.3 header in Ubuntu 14.04

Fix #358