MRPT/mrpt #359: CMake automate download & build of Assimp code

jlblancoc, 2016-10-24:
TO-DO (in a separate branch until stable): 
* [ ] Remove assimp sources from MRPT / otherlibs. It's a huge project and it's a waste of space & time to maintain it here. 
* [ ] Create CMake scripts to automate the download and compilation of ASSIMP from latest GitHub release if the lib is not found in the system (mainly for Windows users).
* [ ] Create patch(es) over upstream sources to ease integration with MRPT. Current changes should be extracted from diffs in `otherlibs/assimp` subdirectory from these commits:
  * https://github.com/MRPT/mrpt/commit/441a11a973c59a43d10a060de385a404a66e551d#diff-5cbde633bb285c2feccca2ad3c62b1fc
  * https://github.com/MRPT/mrpt/commit/e0f741c04b591245359879146f86d642378f3140#diff-2a6e1d877e6c1598c2cc128a978c59a0
  * https://github.com/MRPT/mrpt/commit/3ff27a9145590e56535b34e05c1c88bf50a57ea0#diff-3e9c51ce41f8752d72193fbd8b96c007
  * https://github.com/MRPT/mrpt/commit/494bba7c54777bbf64b91ca4a0b4d9ff4b42a9f7
  * https://github.com/MRPT/mrpt/commit/ba4be990d5e53b7050887891b3b66c9f67e74c72
  * https://github.com/MRPT/mrpt/commit/87c4e6fe9035ebef7d03338e11aea9fd0cdbdd71#diff-3e9c51ce41f8752d72193fbd8b96c007

jlblancoc, 2017-03-02:
Postponed to milestone 2.0.0, where we can require a newer version of CMake and will have stopped supporting Ubuntu Trusty.

jlblancoc, 2017-05-25:
Removing from Milestone 2.0.0, since "Trusty" is already removed, at least from the PPA auto builds list.

bergercookie, 2017-05-25:
@jlblancoc,

>Create CMake scripts to automate the download and compilation of ASSIMP from latest GitHub release if the lib is not found in the system (mainly for Windows users).

Instead of just pulling and working on the latest Assimp release on github, I think it should be more robust to depend on a particular commit/release of the Assimp repo and periodically update that commit ID. This would ensure that nothing breaks in the MRPT compilation process if new non-backwards-compatible code is introduced to Assimp. 

This can be implemented with [Git submodules](https://git-scm.com/book/en/v2/Git-Tools-Submodules) which are designed for maintaining and tracking 3rd party project dependencies.

jlblancoc, 2017-05-25:
Good point! 👍 
Just updated the 2nd task description to reflect that idea.

I liked GitHub submodules, but after working some time with them in another project, I'm now more inclined to prefer leaving that work to CMake, since we avoid having "anything special" in the "git domain"... thoughts?

bergercookie, 2017-05-25:
> I'm now more inclined to prefer leaving that work to CMake, since we avoid having "anything special" in the "git domain"... thoughts?

The "anything special" point is indeed true; unfortunately they do require special attention and can easily break stuff especially in cases that one develops both the outer repo and the submodule itself.

However, if one follows a basic set of rules, they are invaluable at splitting the work in separate repos. I have actually set up my entire [catkin repo](https://github.com/bergercookie/catkin_ws) for the master thesis on git submodules.

Anyway, the CMake solution is pretty neat as well, and also more suitable in this case since MRPT do not hold any other tool as a submodule

jolting, 2017-05-25:
I don't think that is the intent of submodules. Submodules really shouldn't be used to pull in other projects. A submodule should be a way of breaking up a single git repository into smaller pieces(modules), to avoid the issues associated with a large monolithic software repository.

"What happens is that you do a single commit in each submodule that is 
atomic to that *private* copy of that submodule (and nobody will ever see 
it on its own, since you'd not push it out), and then in the supermodule 
you make *another* commit that updates the supermodule to all the changes 
in each submodule." - [Linus Torvald](https://lwn.net/Articles/246381/)

closed 2017-08-13 by jlblancoc

spsancti, 2017-08-16:
Sorry, accidentally referenced this issue.  
Sincerely, Borys.

PR #575 (2017-08-13): Use CMake >=3.1 ExternalProjects for: assimp, octomap, and Eigen
This removes the need to carry embedded copies of those projects within MRPT. System libraries will be used if present; otherwise, they will be downloaded and built automatically as part of the regular build process.

* Closes #574 
* Closes #359