MRPT/mrpt #362: graphslam-engine: error building with AppleClang

jlblancoc, 2016-10-27:
Hi @bergercookie, 

I'll take a look at [this error](https://travis-ci.org/MRPT/mrpt/jobs/171060899#L2977) later myself, but just in case you can fix it more conveniently in your fork before your next PR: 

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/utility:299:15: error: no viable overloaded '='
        first = __p.first;
        ~~~~~ ^ ~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/map:646:15: note: in instantiation of member function 'std::__1::pair<const std::__1::basic_string<char>, int>::operator=' requested here
        {__nc = __v.__cc; return *this;}
              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__tree:1249:35: note: in instantiation of member function 'std::__1::__value_type<const std::__1::basic_string<char>, int>::operator=' requested here
                __cache->__value_ = *__first;
                                  ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__tree:1190:9: note: in instantiation of function template specialization 'std::__1::__tree<std::__1::__value_type<const std::__1::basic_string<char>, int>, std::__1::__map_value_compare<const std::__1::basic_string<char>, std::__1::__value_type<const std::__1::basic_string<char>, int>, std::__1::less<const std::__1::basic_string<char> >, true>, std::__1::allocator<std::__1::__value_type<const std::__1::basic_string<char>, int> > >::__assign_multi<std::__1::__tree_const_iterator<std::__1::__value_type<const std::__1::basic_string<char>, int>, std::__1::__tree_node<std::__1::__value_type<const std::__1::basic_string<char>, int>, void *> *, long> >' requested here
        __assign_multi(__t.begin(), __t.end());
        ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/map:934:21: note: in instantiation of member function 'std::__1::__tree<std::__1::__value_type<const std::__1::basic_string<char>, int>, std::__1::__map_value_compare<const std::__1::basic_string<char>, std::__1::__value_type<const std::__1::basic_string<char>, int>, std::__1::less<const std::__1::basic_string<char> >, true>, std::__1::allocator<std::__1::__value_type<const std::__1::basic_string<char>, int> > >::operator=' requested here
            __tree_ = __m.__tree_;
                    ^
/Users/travis/build/MRPT/mrpt/libs/graphslam/include/mrpt/graphslam/ERD/CICPCriteriaERD_impl.h:401:21: note: in instantiation of member function 'std::__1::map<const std::__1::basic_string<char>, int, std::__1::less<const std::__1::basic_string<char> >, std::__1::allocator<std::__1::pair<const std::__1::basic_string<char>, int> > >::operator=' requested here
        *edge_types_to_num = m_edge_types_to_nums;
                           ^
/Users/travis/build/MRPT/mrpt/libs/graphslam/include/mrpt/graphslam/ERD/CICPCriteriaERD.h:131:3: note: in instantiation of member function 'mrpt::graphslam::deciders::CICPCriteriaERD<mrpt::graphs::CNetworkOfPoses<mrpt::poses::CPosePDFGaussianInf, mrpt::utils::map_traits_stdmap, mrpt::graphs::detail::node_annotations_empty, mrpt::graphs::detail::edge_annotations_empty> >::getEdgesStats' requested here
                CICPCriteriaERD();
                ^
/Users/travis/build/MRPT/mrpt/libs/graphslam/include/mrpt/graphslam/apps_related/TUserOptionsChecker.h:164:14: note: in instantiation of member function 'mrpt::graphslam::deciders::CICPCriteriaERD<mrpt::graphs::CNetworkOfPoses<mrpt::poses::CPosePDFGaussianInf, mrpt::utils::map_traits_stdmap, mrpt::graphs::detail::node_annotations_empty, mrpt::graphs::detail::edge_annotations_empty> >::CICPCriteriaERD' requested here
                return new T;
                           ^
/Users/travis/build/MRPT/mrpt/libs/graphslam/src/TUserOptionsChecker.cpp:64:4: note: in instantiation of function template specialization 'mrpt::graphslam::supplementary::TUserOptionsChecker::createEdgeRegistrationDecider<mrpt::graphslam::deciders::CICPCriteriaERD<mrpt::graphs::CNetworkOfPoses<mrpt::poses::CPosePDFGaussianInf, mrpt::utils::map_traits_stdmap, mrpt::graphs::detail::node_annotations_empty, mrpt::graphs::detail::edge_annotations_empty> > >' requested here
                &createEdgeRegistrationDecider<CICPCriteriaERD<CNetworkOfPoses2DInf> >;
                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:1376:19: note: candidate function not viable: 'this' argument has type 'const std::__1::basic_string<char>', but method is not marked const
    basic_string& operator=(const basic_string& __str);
                  ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:1379:19: note: candidate function not viable: 'this' argument has type 'const std::__1::basic_string<char>', but method is not marked const
    basic_string& operator=(basic_string&& __str)
                  ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:1383:45: note: candidate function not viable: 'this' argument has type 'const std::__1::basic_string<char>', but method is not marked const
    _LIBCPP_INLINE_VISIBILITY basic_string& operator=(const value_type* __s) {return assign(__s);}
                                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:1384:19: note: candidate function not viable: 'this' argument has type 'const std::__1::basic_string<char>', but method is not marked const
    basic_string& operator=(value_type __c);
                  ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:1387:19: note: candidate function not viable: 'this' argument has type 'const std::__1::basic_string<char>', but method is not marked const
    basic_string& operator=(initializer_list<value_type> __il) {return assign(__il.begin(), __il.size());}
                  ^
In file included from /Users/travis/build/MRPT/mrpt/libs/graphslam/src/TUserOptionsChecker.cpp:11:
In file included from /Users/travis/build/MRPT/mrpt/libs/graphslam/include/mrpt/graphslam/apps_related/TUserOptionsChecker.h:23:
In file included from /Users/travis/build/MRPT/mrpt/libs/graphslam/include/mrpt/graphslam/ERD/CLoopCloserERD.h:718:
```

jlblancoc, 2016-10-27:
Probably removing the `const` qualifier to `const string` in:
`std::map<const std::string, int>`
 fixes this, but there might be another better alternative...

bergercookie, 2016-10-27:
I will take a look at it as soon as possible. Thanks for noticing Jose!

commit bbb2f64e (2016-10-30): fix non-assignable key in assoc container

Fixes #362
closed 2016-10-30 by jlblancoc