MRPT/mrpt #408: PTG-based hybrid navigation for articulated vehicles

bellonemauro, 2016-12-26:
Hi @jlblancoc @j3sq,

@j3sq moved here our conversation about the hybrid navigation for articulated vehicles so we are all up to date.

@jlblancoc we have branched/complied the mrpt and we are starting this new implementation, see below the main issues we are dealing with. 

1. Vehicle Kinematic model:

The kinematic model considers a "passive" trailer pulled/pushed by a classical single track Ackermann steering tractor, hence:

control vector : U = [v, w] ---- linear/angular velocity of the tractor    
state vector : X = [x,y, \theta, \alpha] ----- Cartesian position + orientation of the tractor + articulated angle

Forward/backward equations: see docs.

Geometry of the vehicle is :
        
               8----------------7            2--------1
               |                          |            |             |
       T_W |                          5-------4     o      | H_W
               |                         |    L_L    |             |
               9----------------6            3--------0
                      T_L                                H_L

Params:     
   tractor_w  >> width of the tractor
   tractor_l  >>  length of the tractor
   link_l  >> length of the connection link between tractor and trailer 
   trailer_w  >> width of the trailer
   trailer_l  >> length of the trailer

2. PTG implementation:

Here we have the new refactored implmementation of the PTG class: 
https://github.com/MRPT/mrpt/tree/master/libs/nav/include/mrpt/nav/tpspace

The idea is to add a new class there CPTG_artDriving_C inheriting from CParameterizedTrajectoryGenerator, to build the lookup table we are now using an array of classical differential drive lookup tables for each possible articulation angle varying in the range [-30, 30]deg with a resolution of 1deg. 
We will start from the implementation of circular arcs in forward and backward.

3. sensors simulation 

in the current implementation the reactive navigation has a 2D lasers in the center of the vehicle, this should not be correct anymore, we need a proper reconstruction from at least 2 lasers front and back. This must be used to build the Cobstacles.   

4. planner

for now we are using the RRT planner 

Any suggestion will be appreciated

Feliz navidad  @jlblancoc
God jul @j3sq

jlblancoc, 2016-12-26:
Hi @bellonemauro, and Feliz navidad! 👍 

Well, I'd love to see this extension in MRPT, although we must be careful to minimize the impact on existing classes and avoid duplication as much as possible, as you know...

So, step by step on your points:

1) Geometry model: 👍 So basically all it takes is an expanded state vector with \alpha, ok. 
The motion primitive then should remain being the same [CVehicleVelCmd_DiffDriven](https://github.com/MRPT/mrpt/blob/master/libs/kinematics/include/mrpt/kinematics/CVehicleVelCmd_DiffDriven.h). Reusing is good!

2) What did you mean with "reimplemented"? I think there is no need to change anything in the base class for all PTGs, right? If you mean that the new PTG must return a state vector with the extended alpha (right?) then the better would be to define a common template-based virtual base class, not to duplicate it or enforce everyone to use the extended state with alpha (!). 

Please, share here the link to your forked code and I would advice... Also, we can create a branch here in the main MRPT repo where I can start the base template class, you can fork from it, make Pull-requests, etc. ok?

3) Sensors: there is no problem at all with sensors: both reactivenav and RRT uses a **pointcloud**, not a laser scan, to reflect obstacles, so you can use (indeed we routinely do!) two or more lidars as input ;-)

4) RRT is OK.

Ciao!

bellonemauro, 2016-12-27:
Hi,

I have used the wrong word, we will not re-implement, only expand with a new class. 
Fix in the main so we can use it as guideline. 

The branch is in my profile, https://github.com/bellonemauro/mrpt but it is probably better is you create a branch in mrpt and we commit directly there. 

I have also seen that you have a branch with RRT*, which is the level of the implementation in that branch? 

Ciao
M.

jlblancoc, 2017-01-04:
The RRT* branch is still less than a proof of concept... for now you should
go for the RRT solution.

I'll take a look at your branch soon, Mauro...

and Happy New Year, by the way!

bellonemauro, 2017-01-04:
hi, still no commits in the branch, the current implementation is in python and you can see it here https://github.com/j3sq/prrt

Working on the mrpt implementation and documentation.

I wish you a fantastic New Year !!!

jlblancoc, 2018-09-29:
I'll close this old one; will reopen if someone, someday, feels like undertaking the effort of porting it into MRPT ;-)

For the records, here's what I think is the associated 2017 paper with the existing implementation: 
https://ieeexplore.ieee.org/abstract/document/8247633

@bellonemauro : Just if you think it's worth: I added this to the list of 2019 GSoC [ideas page](https://github.com/MRPT/mrpt/wiki/Ideas-page-for-MRPT-Google-Summer-of-Code-2019), if you or some of your team want to be a mentor for that project.

Cheers!

closed 2018-09-29 by jlblancoc