MRPT/mrpt #422: icp-slam-live : no currentPose on raspberry PI

Serveurperso, 2017-01-23:
Hello,

I cloned the mrpt GIT from this repo on a virtual machine and also on a PI Raspbian.
Successfully compiled on two architectures

PC VM :
```
---------------- STEP 441 | RAWLOG ENTRY 0 ----------------
[CMetricMapBuilderICP|DEBUG|08:03:54.6721] processObservation(): extrapolating pose from latest pose and new observation timestamp...
[CMetricMapBuilderICP|WARN |08:03:54.6730] processObservation(): new pose extrapolation failed, using last pose as is.
[CMetricMapBuilderICP|DEBUG|08:03:54.6743] processObservation(): skipping ICP pose correction due to small odometric displacement? : NO
[CMetricMapBuilderICP|DEBUG|08:03:54.6746] processObservation(): matching against gridmap.
[CMetricMapBuilderICP|INFO |08:03:54.6751] processObservation: previousPose=(0.020,0.000,0.04deg)-> currentPose=(0.019,-0.001,-0.12deg)
[CMetricMapBuilderICP|INFO |08:03:54.6754] [CMetricMapBuilderICP]   Fit:82.4% Itr:10 In 0.27ms
[CMetricMapBuilderICP|DEBUG|08:03:54.6756] update map: NO options.enableMapUpdating: YES
Map building executed in 3.582ms
```
It work

ARM :
```
---------------- STEP 53 | RAWLOG ENTRY 0 ----------------
[CMetricMapBuilderICP|DEBUG|08:05:11.5833] processObservation(): extrapolating pose from latest pose and new observation timestamp...
[CMetricMapBuilderICP|WARN |08:05:11.5851] processObservation(): new pose extrapolation failed, using last pose as is.
[CMetricMapBuilderICP|DEBUG|08:05:11.5864] processObservation(): skipping ICP pose correction due to small odometric displacement? : NO
[CMetricMapBuilderICP|DEBUG|08:05:11.5879] processObservation(): matching against gridmap.
[CMetricMapBuilderICP|WARN |08:05:11.5903] Ignoring ICP of low quality: 0
[CMetricMapBuilderICP|DEBUG|08:05:11.5919] update map: NO options.enableMapUpdating: YES
Map building executed in 10.063ms
```
I never get the pose:(

On two machine I use the same sample INI file with disabled 3D scene and rawlog llike this :

```
root@debian:~# diff icp-slam-live.ini ./mrpt/share/mrpt/config_files/icp-slam-live/icp-slam-live_demo_RPLIDAR_gridmap.ini
68c68
< SAVE_3D_SCENE=0
---
> SAVE_3D_SCENE=1
71c71
< SHOW_PROGRESS_3D_REAL_TIME=0
---
> SHOW_PROGRESS_3D_REAL_TIME=1
77c77
< SAVE_RAWLOG = false
---
> SAVE_RAWLOG = true
```

Thanks
Pascal

jlblancoc, 2017-01-23:
Are you using an rplidar? 
Perhaps it's a better idea to start with rawlog-grabber to see if lidar scans are being captured correctly in arm...

Serveurperso, 2017-01-23:
Yes I use live data from rplidar a2
I check this tonight.

Serveurperso, 2017-01-23:
With default sample INI I can look the lidar data. but rawlog-grabber give nothing

```
(root|~) ./rawlog-grabber icpslamlive_dataset__2017-01-23_17h58m43s.rawlog
 rawlog-grabber - Part of the MRPT
 MRPT C++ Library: MRPT 1.5.0 - Sources timestamp: 2017-01-22
-------------------------------------------------------------------
 
Output rawlog filename: dataset_2017-01-23_18h02m33s.rawlog
External image storage: dataset_2017-01-23_18h02m33s_Images
 
 
Press any key to exit program
 
 
Waiting for all threads to close...
(root|~)
```

jlblancoc, 2017-01-23:
Hi,
 I think I couldn't understand you:  do you mean you can see the scans on
the screen but get nothing Into the log file?

Serveurperso, 2017-01-23:
Sorry for my bad english
Your right, there is a strange communication problem like bad power supply.
3D Map worked only once.

closed 2017-01-25 by jlblancoc