MRPT/mrpt #423: TTimeStamp redesign for MRPT 2.0

jlblancoc, 2017-01-23:
* [ ] Transform TTimeStamp into a (POD) `struct`.
* [ ] add methods such as `t.asDouble()`, `t.addInterval()`, `t.asStdChrono()` etc. and deprecate the corresponding global functions.

jolting, 2017-03-07:
I wonder if it would just be better to refactor code that uses TTimeStamp to use directly use std::chrono::time_duration.

asDouble() I believe assumes seconds where as the chrono stuff is much more specific.

If you use `std::chrono::duration_cast<std::chrono::milliseconds, double>` iit is obvious what the double representation is.

Also I really am looking forward to using chrono literals in c++14.
http://en.cppreference.com/w/cpp/chrono/operator%22%22h

jlblancoc, 2017-03-07:
I also thought a lot about using `std::chrono` internally... but in this case I think it will be better to keep our "custom" internal representation, then do as Qt and expose static/non-static methods to convert to/from standard types.

Why?
- Because the standard, AFAIK, does not **ensure** a particular precision/resolution to high resolution timer. In practice, they may have good enough but...
- We still have to serialize / deserialize (including all past datasets in the current format), so at some point we need to convert to something "universal".
- Because the current implementation is very well-tested over many years!

jlblancoc, 2018-08-13:
Closing after #816

closed 2018-08-13 by jlblancoc