MRPT/mrpt #431: CSetOfObjects not registered

wziard, 2017-02-09:
When I try to run the kinect_3d_slam example app I get an assert: 

in libs/opengl/include/mrpt/opengl/CSetOfObjects.h line 150:
DEFINE_SERIALIZABLE_POST_CUSTOM_BASE_LINKAGE(CSetOfObjects, CRenderizable, OPENGL_IMPEXP )

I get a failed assert that the object CSetOfObjects isn't registered. I looked with a debugger and in ithe object registry not all objects are present. It looks like the objects with DEFINE_VIRTUAL_SERIALIZABLE() are registered, but the objects defined with DEFINE_SERIALIZABLE() are not, but I did not (yet) completely confirm this pattern.

Is this a known issue? If it isn't I can look into it in more detail and try to give a more detailed report.

I get rather confused by the myriad of layered #defines though, which is quite awkward to debug.

jlblancoc, 2017-02-09:
Hi @wziard , 

Can you provide more info to help debugging this, please? 
  * MRPT version 
  * Compiler & OS.
  * A minimal example code (ideally, with its CMakeLists.txt file) that gives this error. 

If you are using static linking (not the default), that may be the reason (the explanation on why this happen and how to solve is long, let me know first if this is your case).

closed 2017-02-18 by jlblancoc

wziard, 2017-02-22:
Sorry for the late reaction. The problem was the static linking (probably). I went on to try another (non-mrpt) SLAM demo.

If you still want to know the MRPT version etc I can let you know, but I guess this was the known problem with static linking.