MRPT/mrpt #434: CConfigFile improvements

jlblancoc, 2017-02-13:
To do: 
 - [ ] basic preprocessor: split lines with `\` at end of line.
 - [ ] Unit tests!

GYengera, 2017-02-14:
I am new to mrpt and I assumed that this issue would be something I could solve. Could you please explain what exactly needs to be done?

jlblancoc, 2017-02-16:
Thanks for the offering, @GYengera ! 
I was actually thinking of addressing this in the "short term" (~2 weeks or so, after finishing other topics with higher priority), but this could be indeed a great way for you to get in touch with MRPT so... if you think you can devote time to this in these weeks, it will be fantastic ;-) 

The tasks to do here are: 
 * Create unit tests (read [this](http://www.mrpt.org/libs_tree_layout#6_Unit_testing) and look for examples in existing code) for [CConfigFileMemory](http://reference.mrpt.org/devel/classmrpt_1_1utils_1_1_c_config_file_memory.html). Basic things, as parsing a static `const char *` example configuration file and reading some variables to check if it's parsed right. 
* When done, extend CConfigFileMemory *and* CConfigFile to add a "preprocessor" that allows us to split long config file lines by using the same mechanism than in C/C++ preprocessor: putting a `\` (backslash) *exactly* as the last character in a line. If you are not familiar with this, see [this example](http://stackoverflow.com/questions/10419530/multi-line-preprocessor-macros) (first one I found Googling).

Cheers,

GYengera, 2017-02-19:
@jlblancoc, I have created a unit test for CConfigFileMemory. However, when I run make test, I only see some of the mrpt unit tests are being run. How do I run the new CConfigFileMemory unit test?

jlblancoc, 2017-02-19:
I guess you might need to re-run `cmake .` for CMake to detect your new unit tests and include them in the Makefile

GYengera, 2017-02-23:
The unit test has been run successfully.  I have tested the readString, writeString, getAllKeys and getAllSections functions. How do you want me to send you the code? I will go ahead with the next part now.

jlblancoc, 2017-02-23:
great! :+1:

You can use the Pull request feature of GitHub.

You should have been working until now in a **fork** of MRPT. If not,
just fork it now and copy the changed files to your local copy
("clone") of that fork. Then commit and push to your fork, and
afterwards open a Pull request against this original repo. Make sure
of selecting the `dev-config-files` branch as target branch, instead
of the default `master`.

PR #458 (2017-02-27): CConfigFile basic preprocessor
**Changed apps/libraries:** 
*  mrpt-base: 	CConfigFile now can parse config files with end-of-line backslash to split long strings into several lines. 
* Added cfg file unit tests.

Closes #434
closed 2017-02-27 by jlblancoc