MRPT/mrpt #435: A bug in CPointsMap.cpp?

ygzhangsoya, 2017-02-13:
In [CPointsMap.cpp](https://github.com/MRPT/mrpt/blob/master/libs/maps/src/maps/CPointsMap.cpp)

line 562  `if ( bestMatchForThisMap[it->this_idx] != &(*it) )`: I think it should be `==` instead of `!=`. Is it a bug?

```cpp
   558         //   2) Go again through the list of correspondences and remove those
   559         //       who are not the best one for their corresponding global map.
   560         for (it=_correspondences.begin();it!=_correspondences.end(); ++it)
   561         {
   562             if ( bestMatchForThisMap[it->this_idx] != &(*it) )
   563                 correspondences.push_back( *it );                   // Add to the output
   564         }
```

jlblancoc, 2017-02-13:
Thanks @ygzhangsoya !! 

I'm afraid you're right... please, review my patch above. It basically fixes as you said, by the way updating the coding style to C++11. 

Apparently the bug has been there for... more than 8 years. Please, let us know if you a chance to do some testing with that "unique correspondences filter" enabled, to see if that seems to work better now. 

Cheers

closed 2017-02-13 by jlblancoc

ygzhangsoya, 2017-02-14:
Thanks for the prompt reply. The patch looks good! Thank you for maintaining the library. It's very helpful. 

I think I'll play with it and will let you know how it works now. 

Cheers

ygzhangsoya, 2017-02-15:
Just notice the same bug in line 1204-1208 in [CPointsMap.cpp](https://github.com/MRPT/mrpt/blob/master/libs/maps/src/maps/CPointsMap.cpp). You may want to update it as well.

jlblancoc, 2017-02-16:
Thanks again! I refactored the duplicated code, as can be seen in the commit above.
I tested ICP with `onlyUniqueRobust`=true  and seems to work great.

commit 4667a138 (2017-02-13): fix robust filter for matching points in CPointsMap

Closes #435