MRPT/mrpt #469: (For MRPT 2.0)  Don’t Share Ownership Unnecessarily

jolting, 2017-03-07:
My big concern is that shared pointers imply a memory barrier. This is great for multi-threading, but bad for performance.

The best justification for this I can find is in the O'RIELLY's optimized C++ book[1].

- [ ] Convert shared pointers to references when possible if ownership is not implied
- [ ] Convert shared pointers to dump pointers if references are  not possible and ownership is not implied
- [ ] Convert shared pointers to unique pointers if shared ownership is unnecessary.

[1]https://www.safaribooksonline.com/library/view/optimized-c/9781491922057/

jlblancoc, 2017-03-07:
> My big concern is that shared pointers imply a memory barrier. This is great for multi-threading, but bad for performance.

I never worried about this, actually... well, that's not true: you may find some parts in MRPT where a raw pointer is extracted from a smart pointer whenever it is to be accessed several (>dozens) times within a single function, for example. But I was not aware of the explicit detail on memory barriers...

As far as the changes you propose are mostly "internal" to MRPT classes, I think the changes will be relatively safe and worth doing (but, how to be 100% sure we avoid memory leaks if using raw pointers if we have exceptions?). 

On the other hand, changing anything that is exposed to the final user, so he/she sees "raw pointers" seems to me a bad idea, don't you agree? In any case, he might be responsible of converting the returned shared pointer to a raw pointer, etc. but that would be out of our scope...

PS: Book added to the to-read list, thanks!

jolting, 2017-03-08:
>As far as the changes you propose are mostly "internal" to MRPT classes, I think the changes will be relatively safe and worth doing (but, how to be 100% sure we avoid memory leaks if using raw pointers if we have exceptions?)

You should never *need to* delete a raw pointer.

Something like this `void doSomethingWithObj(Foo *foo)` should imply that `doSomethingWithObj` should be able to throw an exception and not leak. No ownership is inferred.

Generally, raw pointers and references are about the same in most contexts and in general a reference shouldn't imply ownership.

It is usually fairly trivial to convert a raw pointer into a reference or a reference into a raw pointer. I usually prefer a reference over a raw pointer. Mostly because of perfect forwarding. 
The advantages of using `void doSomethingWithObj(const Foo &foo);` over `void doSomethingWithObj(const Foo *foo);` is that the former  could be rewritten as a templated function and it could implement perfect forward where as the later can only handle const pointers.

jlblancoc, 2017-03-08:
ok, I agree with it all!

But I was thinking about a situation like this: sensor driver classes
in mrpt-hwdrivers, which have methods that return smart pointers to
the newly-received observation objects. I thought you meant that this
kind of methods should return raw pointers instead of shared_ptr.

So, if your intention is that passing objects via smart pointers
should be avoided, I agree 100%.
Have you find many such functions in MRPT?
I tend to design (new) APIs based on passing (const)-references, but
the oldest code (~8-10 years ago) may use pointers because back then I
had much less experience and incorrectly believed that pointers were
mandatory when dealing with polymorphic classes (sigh)...

jolting, 2017-03-08:
Specifically I recall fixing it in one place during that giant commit, due to some cyclic dependency on types which made it difficult to use the ::Ptr pointer scheme. The pointer was forward declared before. 
https://github.com/MRPT/mrpt/commit/abf8b3a73b5520ace348f7a87ad6bc137290ac9f#diff-15ee1fca251fe7a45bea7e852479323aL128

jolting, 2017-03-08:
We also have a lot of these
```
T::Ptr t = std::dynamic_pointer_cast<T> (o);
```
This has less overhead
```
T &t = dynamic_cast<T &> (*o);
```
Many cases this replacement can be made, but I chose the former because it was a more direct replacement.

jlblancoc, 2017-03-08:
Ok, I  totally agree with changes like those.

jlblancoc, 2017-03-08:
BTW: Just wondered if you already found `mrpt-performance`... take a look at that app and its parameters. With it, you can compare the performance change between two versions of the library, or if compiled with different flags/compilers/etc.  It's useful if we want to quantify the impact of performance changes. Anyway, profiling still is probably the first thing to look at...

The output for old stable releases, for example: [1](http://reference.mrpt.org/stable/perf-html/index.html) , [2](http://reference.mrpt.org/stable/perf-html/comparison_1.2.0dev-GCC-64bit_vs_1.2.0dev-MSVC11-64bit.html).

Cheers!

jolting, 2017-03-09:
I didn't see a statistically significant difference in runtimes.

jlblancoc, 2017-03-09:
👍 thanks for testing!

PR #1005 (2019-12-07): Better ownership handling
- Avoid using new / delete as much as possible.
- Avoid passing raw pointers, prefer shared_ptr when it makes sense.

Closes #469  (Hopefully!)
commit d98cd6bd (2019-12-07): Better ownership handling:

- Avoid using new / delete as much as possible.
- Avoid passing raw pointers, prefer shared_ptr when it makes sense.

Closes #469  (Hopefully!)
commit 44b84521 (2019-12-09): Better ownership handling:

- Avoid using new / delete as much as possible.
- Avoid passing raw pointers, prefer shared_ptr when it makes sense.

Closes #469  (Hopefully!)
closed 2019-12-07 by jlblancoc