MRPT/mrpt #473: script_detect_gcc.cmake fails with gcc7

jpgr87, 2017-03-12:
[As of gcc 7](https://gcc.gnu.org/ml/gcc-patches/2017-01/msg00851.html), `-dumpversion` just returns "7", not "7.x.y" as required by script_detect_gcc.cmake.  You need to use the new `-dumpfullversion` option to get the full version string.

[This patch](http://pkgs.fedoraproject.org/cgit/rpms/mrpt.git/tree/mrpt-1.4.0-gcc7.patch) tries to work around the issue by trying `-dumpfullversion` if the string returned by `-dumpversion` doesn't have any decimal points.

jlblancoc, 2017-03-13:
Merged... thanks for testing on the edge!

closed 2017-03-13 by jlblancoc

quaiky, 2017-06-23:
this seems not to work for gcc 5.3 which also returns only "5" for -dumpversion but does not know the  -dumpfullversion command. 
```bash
# g++ -dumpversion
5
# g++ -dumpfullversion
g++: fatal error: no input files
compilation terminated.
# g++ --version
g++ (SUSE Linux) 5.3.1 20160301 [gcc-5-branch revision 233849]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jlblancoc, 2017-08-03:
Thanks @quaiky !

After some research, I found most of this code was useless: CMake already provides a safer way to determine the compiler version via `${CMAKE_CXX_COMPILER_VERSION}`. 
The fix to support any compiler version is on commit https://github.com/MRPT/mrpt/commit/2dd4531ab95a114dfb70e3e4663465ea71bb13fc

Cheers!

commit 3e3c86be (2017-03-13): Fix GCC version detection in GCC 7

(Closes #473)