MRPT/mrpt #478: OMPL integration?

olzhas, 2017-03-14:
As an idea what do you think about OMPL integration into MRPT?
"OMPL, the Open Motion Planning Library, consists of many state-of-the-art sampling-based motion planning algorithms." http://ompl.kavrakilab.org/

jlblancoc, 2017-03-16:
Hi @olzhas !

Naturally, we knew that great library, and at some point in the past it was considered whether it makes sense or not to integrate it into MRPT in some way... 

My conclusion was that OMPL is orthogonal to MRPT in aims and (almost 100%) in scope. Yes, MRPT has some basic RRT vehicle planners and alike, but OMPL is *far* more general (and complex!) so it can handle arbitrary kinematic chains, etc.

So, if both MRPT and OMPL are required in an application, they could be easily integrated at the user application level... 

In short: I'm happy with MRPT aiming for offering "simple" versions of *some* path planners, but if the full pack is required, users should switch to OMPL or other alternatives ;-)

Cheers

closed 2017-03-16 by jlblancoc

olzhas, 2017-03-16:
@jlblancoc thank you for your comment, I was thinking what to do for GSoC2017, and I thought that it would be a good idea to provide some bindings to this library.

jlblancoc, 2017-03-16:
I see... 

It could still be interesting, but probably better as a separate sub-project / new library, which might use both MRPT & OMPL as usual via CMake FIND_PACKAGE() and such... 

Sometimes I'm concerned with the size of MRPT and the effort of maintaining it... :-)