MRPT/mrpt #482: Add debugging symbols to mrpt library for external project use - source compilation

bergercookie, 2017-03-17:
@jlblancoc,  loosely based on @alexge233's issue [here](https://github.com/MRPT/mrpt/issues/479), I encountered the same error log when I tried adding debugging symbols to the mrpt-graphslam library.

I compiled mrpt source using the CMAKE_BUILD_TYPE=Debug flag so that I could use the debugger in the mrpt_slam/mrpt_graphslam_2d application (which apparently made use of the mrpt-graphslam library).

jolting, 2017-03-17:
https://github.com/MRPT/mrpt/issues/211

I propose don't append -dbg on Linux.

bergercookie, 2017-03-17:
I think I've come to this issue again. However if you compile with the DONT_USE_DEBUG_LIBS flag, the debugger can't inspect any part of the program related to the MRPT library that you are using (stack trace, lines, variable contents etc). Isn't this the case? I was hoping for a way of getting the line that the program crashes, even if that is in mrpt-graphslam lines..

jolting, 2017-03-17:
No. If compiled with debug symbols it will use them. 

The dbg package just installs it in a place where gdb can find the symbols. They are stripped out of the library and stuck in a seperate folder, so the library name should stay the same regardless of using debug symbols.

jlblancoc, 2017-03-17:
> #211
>
> I propose don't append -dbg on Linux.

I think it's time to surrender... you're right: this convention makes
sense and is really useful in Windows, but seems a permanent source of
confusion in Linux.

jlblancoc, 2017-03-18:
Closed after merge of #484

closed 2017-03-18 by jlblancoc

jolting, 2017-03-18:
I think windows would be better if it had some rpath analog. It's very useful to have local builds around and embed rpath in the binary, so you don't have to manually specify library search paths. 

Windows is a bit unruly. I've had to once or twice manually gather dlls to run my apps.it makes sense on windows to add -dbg for this reason.