MRPT/mrpt #490: Add `alloca`-compatible mem allocator

jlblancoc, 2017-03-22:
For usage within STL containers and such whose scope is limited and are to hold temporary objects only.

jolting, 2017-04-11:
This is worth a read:
http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2013/n3662
https://gcc.gnu.org/ml/libstdc++/2013-06/msg00045.html

I don't think the proposal went anywhere due to the challenges of implementing it.

jlblancoc, 2017-04-11:
Ah!! I forgot coming back here to comment, sorry! 

Last week I spent some time trying to implement this... just to realize that, as you saw, it seems *impossible*, at least, extremely challenging if we want it to be portable. 

The main weakness is that ::alloca() **must be called** from the same stack frame than the user code that will use that memory; so it must be done via macros (impossible for STL allocator), or via 100% inline code... and who ensures the latter? (that's basically what your second link tells!)

So, I'm closing this one...

BTW: At least for `std::map` we have [this](https://github.com/MRPT/mrpt/blob/master/libs/base/include/mrpt/utils/map_as_vector.h) which is a significant improvement in number of calls to the mem manager. Similar ideas could be extended to other containers, perhaps via a custom `allocator` instead of my intrusive approach in `map_as_vector<>`.

closed 2017-04-11 by jlblancoc