MRPT/mrpt #496: (For MRPT 2.0) Is it possible to use PIMPL-like idiom for Eigen?

jlblancoc, 2017-03-30:
(cc: @jolting ) I noticed a large part of compilation time slow-down comes from the use of extensive, header-only libs, such as Eigen.

I wonder if there might exist an **efficient** solution that fulfills all:
- Allows us to get Eigen `#include`s out of MRPT headers, 
- But does **not** enforce using dynamic memory allocation.

Perhaps one of the value `variant` containers may do the work? 

I thought of using the PIMPL idiom, then allowing users that really need to use all Eigen methods to do the #include on their side and call a templatized getter method, or something alike.
Obviously, this is not acceptable since replacing all statically allocated matrices (they are inside all CPose3D, etc. classes) with dynamic memory is a shot in the shoe!

I think it's not possible, but just wanted to start this conversation in case there is some remote possibility of doing this without a huge efficiency cost...

jolting, 2017-03-30:
Maybe something like the PCL methodology?

They instantiate template types in a CPP file.
https://github.com/PointCloudLibrary/pcl/blob/master/filters/src/approximate_voxel_grid.cpp

They implement it in a impl header
https://github.com/PointCloudLibrary/pcl/blob/master/filters/include/pcl/filters/impl/approximate_voxel_grid.hpp

Typically you don't need the impl header(which takes the longest to compile) unless you need to instantiate versions.
https://github.com/PointCloudLibrary/pcl/blob/master/filters/include/pcl/filters/approximate_voxel_grid.h

jlblancoc, 2017-03-30:
good! 👍 

Will check if Eigen headers are already in a suitable form...

jolting, 2017-03-30:
I don't believe they are. This will make deriving from Eigen types almost impossible. Possibly using composition instead of derivation will be the only solution. Basically, only use Eigen in the impl header.

jlblancoc, 2017-04-02:
As a reference to think about, here is the output from `gcc -ftime-report` for: 

1) A "simple" file without any heavy use of templates (0.34 s to compile)

```
[ 25%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/utils/CServerTCPSocket_common.cpp.o

Execution times (seconds)
 phase setup             :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall    1383 kB ( 5%) ggc
 phase parsing           :   0.09 (26%) usr   0.03 (38%) sys   0.12 (28%) wall   11235 kB (37%) ggc
 phase lang. deferred    :   0.08 (24%) usr   0.03 (38%) sys   0.11 (26%) wall   11698 kB (39%) ggc
 phase opt and generate  :   0.17 (50%) usr   0.02 (25%) sys   0.19 (44%) wall    5822 kB (19%) ggc
 |name lookup            :   0.03 ( 9%) usr   0.00 ( 0%) sys   0.05 (12%) wall    2738 kB ( 9%) ggc
 |overload resolution    :   0.06 (18%) usr   0.02 (25%) sys   0.08 (19%) wall    8013 kB (27%) ggc
 callgraph construction  :   0.02 ( 6%) usr   0.00 ( 0%) sys   0.04 ( 9%) wall    1419 kB ( 5%) ggc
 callgraph optimization  :   0.00 ( 0%) usr   0.01 (13%) sys   0.01 ( 2%) wall     151 kB ( 1%) ggc
 ipa SRA                 :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall     263 kB ( 1%) ggc
 df live regs            :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall       0 kB ( 0%) ggc
 alias stmt walking      :   0.00 ( 0%) usr   0.01 (12%) sys   0.00 ( 0%) wall      14 kB ( 0%) ggc
 preprocessing           :   0.02 ( 6%) usr   0.01 (12%) sys   0.03 ( 7%) wall     386 kB ( 1%) ggc
 parser (global)         :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.02 ( 5%) wall    2844 kB ( 9%) ggc
 parser struct body      :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall    1271 kB ( 4%) ggc
 parser function body    :   0.01 ( 3%) usr   0.01 (12%) sys   0.01 ( 2%) wall    1241 kB ( 4%) ggc
 parser inl. func. body  :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall      61 kB ( 0%) ggc
 parser inl. meth. body  :   0.02 ( 6%) usr   0.01 (12%) sys   0.03 ( 7%) wall    1998 kB ( 7%) ggc
 template instantiation  :   0.10 (29%) usr   0.01 (13%) sys   0.14 (33%) wall   14901 kB (49%) ggc
 integration             :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall     598 kB ( 2%) ggc
 tree gimplify           :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall     293 kB ( 1%) ggc
 tree CFG cleanup        :   0.02 ( 6%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall       4 kB ( 0%) ggc
 tree PTA                :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall      31 kB ( 0%) ggc
 dominator optimization  :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall      23 kB ( 0%) ggc
 tree CCP                :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall      21 kB ( 0%) ggc
 tree PRE                :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall      22 kB ( 0%) ggc
 tree FRE                :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall      56 kB ( 0%) ggc
 CSE                     :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall       9 kB ( 0%) ggc
 dead store elim1        :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall      15 kB ( 0%) ggc
 CPROP                   :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall      22 kB ( 0%) ggc
 combiner                :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall      58 kB ( 0%) ggc
 integrated RA           :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall     314 kB ( 1%) ggc
 LRA non-specific        :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall       6 kB ( 0%) ggc
 scheduling 2            :   0.01 ( 3%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall      17 kB ( 0%) ggc
 rest of compilation     :   0.02 ( 6%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall      39 kB ( 0%) ggc
 unaccounted todo        :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 2%) wall       0 kB ( 0%) ggc
 TOTAL                 :   0.34             0.08             0.43              30167 kB
```

2) A file making extensive use of templates (Eigen, mostly), taking 16.6 s to build:

```
[  9%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/random/RandomGenerator.cpp.o

Execution times (seconds)
 phase setup             :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall    1383 kB ( 0%) ggc
 phase parsing           :   0.05 ( 0%) usr   0.00 ( 0%) sys   0.06 ( 0%) wall    4374 kB ( 0%) ggc
 phase lang. deferred    :   5.70 (34%) usr   0.99 (50%) sys   6.69 (36%) wall  620261 kB (63%) ggc
 phase opt and generate  :  10.85 (65%) usr   0.98 (50%) sys  11.83 (64%) wall  355111 kB (36%) ggc
 |name lookup            :   0.64 ( 4%) usr   0.22 (11%) sys   0.95 ( 5%) wall   48747 kB ( 5%) ggc
 |overload resolution    :   3.40 (20%) usr   0.66 (34%) sys   4.15 (22%) wall  441614 kB (45%) ggc
 garbage collection      :   1.37 ( 8%) usr   0.00 ( 0%) sys   1.39 ( 7%) wall       0 kB ( 0%) ggc
 dump files              :   0.13 ( 1%) usr   0.05 ( 3%) sys   0.17 ( 1%) wall       0 kB ( 0%) ggc
 PCH preprocessor state restore:   0.01 ( 0%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall       0 kB ( 0%) ggc
 callgraph construction  :   0.35 ( 2%) usr   0.03 ( 2%) sys   0.36 ( 2%) wall   12859 kB ( 1%) ggc
 callgraph optimization  :   0.27 ( 2%) usr   0.06 ( 3%) sys   0.39 ( 2%) wall   13989 kB ( 1%) ggc
 ipa dead code removal   :   0.01 ( 0%) usr   0.00 ( 0%) sys   0.00 ( 0%) wall       0 kB ( 0%) ggc
 ipa cp                  :   0.00 ( 0%) usr   0.00 ( 0%) sys   0.01 ( 0%) wall    1417 kB ( 0%) ggc
[... cut ...]
Just a crazy idea: might it be useful to have Eigen refactored into declaration headers and implementation files, which might be explicitly instantiated? It might imply a small performance loss, but it should be measured to evaluate if it's acceptable. 
I know Eigen is extremely flexible and large, but this idea could be applied to just the most commonly-used instantiations, then linked against in most parts instead of telling the compiler to generate the same code over and over again...

This could be a big workload on itself and no one has spare time to work on this, but... just wanted to share my thoughts!

jolting, 2017-04-02:
My though was refactoring CMatrix and associated types into a c++14 implementation with allows explicit conversions to Eigen. I think this would be a good start.

C++11 arrays can be stack allocated, which should improve performance. 

Ideally, the c++14 implementation should focus on constexpr, so that the compiler can do as many optimization as compile time as possible.

http://codereview.stackexchange.com/questions/136541/simple-matrix-class-c14

jlblancoc, 2017-04-03:
mmm... it feels like rolling back the decision back taken more than 5 years ago when moving to Eigen from custom matrix classes!

But it's worth considering for 2.0: it might be the simplest way of achieving small, simple templates for storing matrices, then only including `<Eigen/...>` where really necessary. 

That idea would work if we draw a clear border for what operations lie on "our side", with the rest being for Eigen. An idea is to just implement the most basic storage. 

By the way: [`Eigen::Map<>`](https://eigen.tuxfamily.org/dox/classEigen_1_1Map.html) is a way to go for such a custom matrix storage, e.g. (just a tentative idea)

```
template <...>
class matrix 
{
...
auto toEigen() {
  return Eigen::Map<Eigen::Matrix<...> > (this->m_storage);
}
...
};
```

(PS: just edited this issue title)

jlblancoc, 2017-04-14:
I just realized now that my last code snippet didn't make sense, since it implies having `#include`d all Eigen headers so `Eigen::Map<>` and such are defined, and that's exactly what we want to avoid! :-)

This one should be better, as it will work only from user sources where Eigen headers are included.
This time the code is tested (in MSVC 2015):

HEADER:
```
#include <array>

// Frwd decls for Matrix:
namespace Eigen {
	template<typename PlainObjectType, int MapOptions, typename StrideType> class Map;
	template<int Value> class InnerStride;
}

template <typename T, int ROWS, int COLS>
struct Matrix
{
	template <typename EIGEN_MATRIX, typename EIGEN_MAP = Eigen::Map<EIGEN_MATRIX, 1 /*16 align*/, Eigen::InnerStride<1> /* row major */> >
	EIGEN_MAP asEigen() {
		return EIGEN_MAP(&m_matrix[0], ROWS, COLS);
	}

private:
	alignas(16) std::array<T, ROWS*COLS> m_matrix;
};
```

SOURCE:
```
#include <mrpt/utils/types_math.h>  // Eigen hdrs

Matrix<double, 4, 4> M;

auto em = M.asEigen< Eigen::Matrix4d >();
em.setIdentity();
std::cout << em << std::endl;
```

jlblancoc, 2017-05-28:
Updated issue first comment to reflect a couple of TO-DO tasks related to this discussion.

jlblancoc, 2019-03-29:
WIP for this issue is branch: [matrix-hidden-eigen](https://github.com/MRPT/mrpt/tree/matrix-hidden-eigen)

PR #916 (2019-05-24): WIP: Big refactoring of matrix classes (a.k.a. "hidden Eigen")
This is the largest change in MRPT math classes since the introduction of Eigen years ago. 
Including Eigen headers makes compilation of user apps slow, so the new approach is to define vector & matrix containers with a minimum of Algebra functionality, but much faster to compile. The containers can still be `Eigen::Map`ed to Eigen classes as needed. 

At present, rebuilding all mrpt libs is 25% faster after this change.

Progress: 
* [X] Update all mrpt libs.
* [x] Update mrpt apps.
* [x] Double check that all unit tests pass.
* [ [... cut]
closed 2019-05-24 by jlblancoc