MRPT/mrpt #502: graphslam-engine: Make SLAM evalution metric independent of CGraphSlamEngine code

bergercookie, 2017-03-30:
CGraphSlamEngine utilizes a benchmarking criterion initially proposed in [1]. This is a rather easy to grasp, but fairly handy metric for evaluating the performance of a SLAM algorithm. It utilizes only the robot estimated trajectory and compares some of its poses with the corresponding poses of the ground-truth trajectory. Thus it is independent of the SLAM algorithm itself (i.e. not only applicable to grpah-slam). This should be reformatted into its own separate class (e.g. mrpt::slam::CEvaluationMetric) so it can be used by any other SLAM algorithm in MRPT.

[1] [A Comparison of SLAM Algorithms Based on a Graph of Relations](http://europa.informatik.uni-freiburg.de/files/burgard09iros.pdf)

Proposed tag: Enhancement
Proposed milestone: GraphSLAM engine improvements

jlblancoc, 2017-03-30:
Great idea! And good engineering work, encouraging code reuse by factorizing it out 👍 

Just curious: how does that method compare to Sturm's [Python scripts](http://vision.in.tum.de/data/datasets/rgbd-dataset/tools#evaluation) ([paper](http://ais.informatik.uni-freiburg.de/publications/papers/sturm12iros.pdf)) for a pretty similar task?

bergercookie, 2017-03-31:
Hmm, that's interesting! As the paper suggests in VII.A the
evaluate_rpe.py script
(relative pose error) implements, in an offline fashion, exactly the
strategy that I am suggesting!

jlblancoc, 2017-08-20:
Any news on this one, @bergercookie ?? Should we close the ticket if no action is to be taken?

BTW: I organized all pending graphslam issues as a "project" instead of a milestone. See [it here](https://github.com/MRPT/mrpt/projects/1).

bergercookie, 2017-08-20:
> Any news on this one, @bergercookie <https://github.com/bergercookie> ??
> Should we close the ticket if no action is to be taken?

Should be quite useful, but currently I have other tasks to attend to. I
think it would be better to keep it around in the to-do list of the
GraphSlam engine project. Do close though it if you think it's better! :-)

> BTW: I organized all pending graphslam issues as a "project" instead of a
> milestone. See it here <https://github.com/MRPT/mrpt/projects/1>.

Nice!

>
>
> <https://github.com/MRPT/mrpt/issues/502#issuecomment-323555431>, or mute
> the thread
-- 
Nikos Koukis
School of Mechanical Engineering
National Technical University of Athens
nickkouk@gmail.com
+30 6985827375

jlblancoc, 2022-09-21:
I'm afraid this one is outdated by now... :-( 
Closing it...

(Some python3 packages exist for this nowadays)

closed 2022-09-21 by jlblancoc