MRPT/mrpt #514: Implement SOFT features in mrpt-vision

jlblancoc, 2017-04-30:
* [Paper reference]http://ieeexplore.ieee.org/abstract/document/7324219/). Seems to outperform many state-of-the-art alternatives. 
* Example MATLAB implementation [here](https://github.com/Mayankm96/Stereo-Odometry-SOFT)

srihegde, 2017-07-22:
I am interested in implementing this feature. Can you give me some cue about where to start from?
Thanks!

jlblancoc, 2017-07-24:
Thanks for the initiative! I'll pass you soon a few pointers on how to get
started.

srihegde, 2017-07-25:
Since this is a new feature detector for this repo I am thinking of creating a new file altogether, like `mrpt/libs/vision/src/CFeatureExtraction_SOFT.cpp` and its required utilities in a new folder `mrpt/libs/vision/src/SOFT_utils` for better organization.
Let me know if you have any corrections here.

jlblancoc, 2017-07-26:
Yes, that's perfect.

Please be sure of carefully reading this:
https://github.com/MRPT/mrpt/blob/master/.github/CONTRIBUTING.md

Cheers!

srihegde, 2017-08-01:
I started with implementation of the utilities first. So I wanted to know if any of the following standard Matlab APIs are implemented in MRPT -

- [detectMinEigenFeatures](https://in.mathworks.com/help/vision/ref/detectmineigenfeatures.html): Detecting Shi-Tomasi features in image
- [extractFeatures](https://in.mathworks.com/help/vision/ref/extractfeatures.html): Extracting detected features.
- [matchFeatures](https://in.mathworks.com/help/vision/ref/matchfeatures.html)
- [KLT Tracker (Kanade-Lucas-Tomasi)](https://in.mathworks.com/help/vision/ref/vision.pointtracker-class.html)
- [showMatchedFeatures](https://in.mathworks.com/help/vision/ref/showmatchedfeatures.html): For visualization of matched features

Thanks.

jlblancoc, 2017-08-21:
Sorry for the delay but as you might have seen... we have too many interesting developing fronts right now!

Answering your doubts:

> detectMinEigenFeatures: Detecting Shi-Tomasi features in image

[`mrpt::vision::CFeatureExtraction::extractFeaturesKLT()`](http://mrpt.ual.es/reference/devel/classmrpt_1_1vision_1_1_c_feature_extraction.html) does the job if you set `options.featsType` to `featKLT`.

> extractFeatures: Extracting detected features.

The same class linked above has a `computeDescriptors()` method with a number of options. Refer to the Doxygen docs in the link.

> matchFeatures 

Yes: [mrpt::vision::matchFeatures](http://mrpt.ual.es/reference/devel/group__mrpt__vision__grp.html#gad073bd2b1235df6a75e73a72df701204)

> KLT Tracker (Kanade-Lucas-Tomasi)

Yes: [mrpt::vision::CFeatureTracker_KL](http://mrpt.ual.es/reference/devel/structmrpt_1_1vision_1_1_c_feature_tracker___k_l.html)

> showMatchedFeatures: For visualization of matched features

Not as a single function, as far as I remember (@famoreno ??), but you could do:

```cpp
CFeatureList list1,list2;
// ...
mrpt::gui::CDisplayWindow win1, win2;
win1.showImageAndPoints( img1, list1 );
win2.showImageAndPoints( img2, list2 );
```

srihegde, 2017-08-22:
Thanks! That was really helpful.

Doodle1106, 2020-05-14:
has this feature been implemented?

jlblancoc, 2020-05-14:
Nope... if someone wants to contribute it, fork+pull-requests are welcome! ;-)

closed 2026-10-02 by jlblancoc