MRPT/mrpt #519: Linking error with assimp-mrpt library

bergercookie, 2017-05-24:
I came across the following linking error when trying to compile MRPT.

```
/usr/bin/ld: cannot find -lassimp-mrpt
collect2: error: ld returned 1 exit status
make[3]: *** [lib/libmrpt-maps.so.1.5.0] Error 1
make[2]: *** [libs/maps/CMakeFiles/mrpt-maps.dir/all] Error 2
make[1]: *** [apps/graphslam-engine/CMakeFiles/graphslam-engine.dir/rule] Error 2
make: *** [graphslam-engine] Error 2

```

Tinkering with the produced make files and by-products, the link.txt responsible for providing the linking command is wrong. In the following command the mrpt build directory is /home/berger/mrpt_build, and `assimp-mrpt.so` is successfully produced in the `/home/berger/mrpt_build/lib` directory. However, the corresponding `-L` flag in c++ is `-L/home/berger/lib` Having tested this a second time, with the build directory being located in e.g. /home/berger/X/Y make produces `-L/home/berger/X/lib` so it essentially skips the last directory in the path.

```
/usr/bin/c++ -fPIC -I /usr/include/wx-3.0 -I /usr/lib/x86_64-linux-gnu/wx/include/gtk2-unicode-3.0 -pthread -Wno-deprecated -Wreturn-type -Wextra -Wtype-limits -Wcast-align -Wparentheses -Wno-unused-parameter -Wall -Wno-long-long -Wno-variadic-macros -Wno-write-strings -std=c++11 -O3 -mtune=native -msse2 -funroll-loops -mfpmath=sse -msse3 -mssse3 -msse4.1 -msse4.2 -O3 -DNDEBUG -shared -Wl,-soname,libmrpt-opengl.so.1.5 -o ../../lib/libmrpt-opengl.so.1.5.0 CMakeFiles/mrpt-opengl.dir/src/lib3ds/quat.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/light.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/tracks.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/mesh.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/viewport.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/material.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/file.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/camera.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/vector.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/ease.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/io.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/background.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/chunk.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/matrix.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/tcb.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/atmosphere.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/shadow.c.o CMakeFiles/mrpt-opengl.dir/src/lib3ds/node.c.o CMakeFiles/mrpt-opengl.dir/src/CSetOfTexturedTriangles.cpp.o CMakeFiles/mrpt-opengl.dir/src/CText.cpp.o CMakeFiles/mrpt-opengl.dir/src/COpenGLScene.cpp.o CMakeFiles/mrpt-opengl.dir/src/CEllipsoid.cpp.o CMakeFiles/mrpt-opengl.dir/src/CFBORender.cpp.o CMakeFiles/mrpt-opengl.dir/src/CVectorField2D.cpp.o CMakeFiles/mrpt-opengl.dir/src/CLight.cpp.o CMakeFiles/mrpt-opengl.dir/src/CFrustum.cpp.o CMakeFiles/mrpt-opengl.dir/src/CVectorField3D.cpp.o CMakeFiles/mrpt-opengl.dir/src/CBox.cpp.o CMakeFiles/mrpt-opengl.dir/src/COpenGLViewport.cpp.o CMakeFiles/mrpt-opengl.dir/src/CColorBar.cpp.o CMakeFiles/mrpt-opengl.dir/src/CDisk.cpp.o CMakeFiles/mrpt-opengl.dir/src/CSetOfTriangles.cpp.o CMakeFiles/mrpt-opengl.dir/src/CGeneralizedEllipsoidTemplate.cpp.o CMakeFiles/mrpt-opengl.dir/src/CEllipsoidInverseDepth3D.cpp.o CMakeFiles/mrpt-opengl.dir/src/CPolyhedron.cpp.o CMakeFiles/mrpt-opengl.dir/src/CTexturedPlane.cpp.o CMakeFiles/mrpt-opengl.dir/src/CGeneralizedCylinder.cpp.o CMakeFiles/mrpt-opengl.dir/src/CGridPlaneXZ.cpp.o CMakeFiles/mrpt-opengl.dir/src/CEllipsoidRangeBearing2D.cpp.o CMakeFiles/mrpt-opengl.dir/src/CMesh.cpp.o CMakeFiles/mrpt-opengl.dir/src/CCylinder.cpp.o CMakeFiles/mrpt-opengl.dir/src/CText3D.cpp.o CMakeFiles/mrpt-opengl.dir/src/CTextMessageCapable.cpp.o CMakeFiles/mrpt-opengl.dir/src/CSetOfLines.cpp.o CMakeFiles/mrpt-opengl.dir/src/CPointCloudColoured.cpp.o CMakeFiles/mrpt-opengl.dir/src/CAssimpModel.cpp.o CMakeFiles/mrpt-opengl.dir/src/CMesh3D.cpp.o CMakeFiles/mrpt-opengl.dir/src/C3DSScene.cpp.o CMakeFiles/mrpt-opengl.dir/src/CTexturedObject.cpp.o CMakeFiles/mrpt-opengl.dir/src/COctoMapVoxels.cpp.o CMakeFiles/mrpt-opengl.dir/src/CEllipsoidInverseDepth2D.cpp.o CMakeFiles/mrpt-opengl.dir/src/CArrow.cpp.o CMakeFiles/mrpt-opengl.dir/src/registerAllClasses.cpp.o CMakeFiles/mrpt-opengl.dir/src/CSetOfObjects.cpp.o CMakeFiles/mrpt-opengl.dir/src/COpenGLStandardObject.cpp.o CMakeFiles/mrpt-opengl.dir/src/pose_pdfs.cpp.o CMakeFiles/mrpt-opengl.dir/src/CRenderizableDisplayList.cpp.o CMakeFiles/mrpt-opengl.dir/src/CSimpleLine.cpp.o CMakeFiles/mrpt-opengl.dir/src/CMeshFast.cpp.o CMakeFiles/mrpt-opengl.dir/src/CRenderizable.cpp.o CMakeFiles/mrpt-opengl.dir/src/gl_utils.cpp.o CMakeFiles/mrpt-opengl.dir/src/CSphere.cpp.o CMakeFiles/mrpt-opengl.dir/src/CPointCloud.cpp.o CMakeFiles/mrpt-opengl.dir/src/CCamera.cpp.o CMakeFiles/mrpt-opengl.dir/src/StockObjects.cpp.o CMakeFiles/mrpt-opengl.dir/src/opengl-precomp.cpp.o CMakeFiles/mrpt-opengl.dir/src/CGridPlaneXY.cpp.o CMakeFiles/mrpt-opengl.dir/src/CAxis.cpp.o -L/home/berger/lib -lrt ../../lib/libmrpt-base.so.1.5.0 -lGL -lGLU -lglut -lassimp-mrpt -lrt /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.8 -lamd -lccolamd -lcolamd -lcholmod -lspqr -lldl -lbtf -lklu -lcxsparse -lumfpack -lblas -llapack -L/usr/lib/x86_64-linux-gnu -pthread -lwx_baseu-3.0 -lwx_gtk2u_core-3.0 -lwx_gtk2u_gl-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_aui-3.0 -lwx_gtk2u_html-3.0 -lz -ljpeg -Wl,-rpath,/home/berger/lib:/home/berger/mrpt_build/lib
```

The linking error occurs also in the linking of mrpt-opengl, mrpt-graphs and mrpt-gui, mrpt-vision libraries. 
Manually correcting the path in all corresponding link.txt files solves the problem and the compilation finishes successfully. 

Attached are files containing the failing command, OS/installed software information as well as the CMakeCache of the build directory at the time of compilation.

P.S. Current issue is not related with the ongoing PR.

[CMakeCache.txt](https://github.com/MRPT/mrpt/files/1027107/CMakeCache.txt)
[failing_make_command.txt](https://github.com/MRPT/mrpt/files/1027112/failing_make_command.txt)
[software_versions.txt](https://github.com/MRPT/mrpt/files/1027113/software_versions.txt)

bergercookie, 2017-05-24:
**Update:** 

Deactivating assimp compilation completely seems to fix the problem.

jlblancoc, 2017-05-24:
Have you tried starting from an empty build cmake directory? Just in case...

bergercookie, 2017-05-24:
Yes. Didn't  work.

jolting, 2017-05-24:
Travis installs libassimp-dev

Isn't this failing when it tries to use the internal version in otherlibs?

I'm in favor of removing assimp from otherlib, which would force users to use an external version if they want to use assimp. Is there a good reason to use the internal one?

jolting, 2017-05-24:
I think this was the suggestion in #359

jlblancoc, 2017-05-25:
> I'm in favor of removing assimp from otherlib, which would force users to use an external version if they want to use assimp. Is there a good reason to use the internal one?

Windows users. In particular, lazy Windows users. Many potential users shall fall in this category ;-)

> I think this was the suggestion in #359

Correct! 
I just revised #359 and removed its "milestone 2.0.0" tag, which means that it could be addressed now. 

Anyone wanting to take an eye on #359 ? Perhaps someone from the GSOC2017 gang (cc: @shivangag @pranjal-rai @spsancti @raghavendersahdev @LisGein ) wanting to gain a good insight into MRPT CMake scripts? ;-)

spsancti, 2017-05-26:
@jlblancoc I have a very little knowledge on CMake, however, I can try this on weekend :)

jolting, 2017-05-26:
@spsancti Learning CMake is an valuable skill. Its worth spending time learning how to do it properly. There are a fair number of conventions that are important to know and understand, but that are often neglected. MRPT is a good place to learn how to use it effectively.

jlblancoc, 2017-06-04:
I think we can close this one and leave the open working line #359 , ok?

closed 2017-06-04 by jlblancoc